dotintent/react-native-ble-plx #165: ble-plx affecting the network connectivity and fetch calls

vldgeorgiev, 2017-10-30:
There is a strange problem that I have, not sure how to describe it best.
In short, it seems that functions to read/write a characteristic also affect the network connectivity of the app. 

I use these functions
```
bleManager.monitorCharacteristicForDevice(deviceId, 'service', 'characteristic', handleResponse, null);
```
and 
```
writeCharacteristicWithoutResponseForDevice()
```

The `handleResponse` function parses the characteristic value and dispatches some async actions(with redux-thunk) that use `fetch`.

Very often these fetch calls fail with a "Network request failed". Also, afterwards I get the error that the app can't connect to the React debugger due to network problems.
The error happens more if I place a breakpoint somewhere in the code and stop at it.

Does that issue ring a bell for someone? It is quite strange to me why the BLE functions would affect the async actions.

Running on Android 6.0, RN 0.42, react-native-ble-plx 0.6.4

alanyong91, 2017-12-16:
I got this error: 

`writeCharacteristicWithoutResponseForDevice() is not a function`

Any idea why?

Cierpliwy, 2018-01-05:
How is it related to this issue @alanyong91 ?

FilipeAraujo, 2018-01-31:
@vldgeorgiev did you fixed this issue? or at least understand why that happens? I'm having some similar issues.

vldgeorgiev, 2018-01-31:
@FilipeAraujo Unfortunately no, I haven't fixed it and not sure why it happens. In my case it only happens while debugging, not in the release build, so I managed to work around it. But even in debug only it is annoying for sure.

closed 2019-03-23 by stale[bot]