dotintent/react-native-ble-plx #169: Support for eddystone urls

Jowin, 2017-11-01:
Is there a way to get eddystone beacon details while scanning ? Fields like Instance Id seems to be missing while scanning.

Cierpliwy, 2017-11-30:
Currently we can only send following fields of advertisement data on iOS:
https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerdelegate/advertisement_data_retrieval_keys
On Android we have all advertisement data bytes available, but I decided to parse only above to keep compability with iOS.

Jowin, 2017-12-11:
Thanks for the revert, I was unable to find any documentation with regard to the advertisement data. Could you point to some part of the documentation for the library where the advertisement data is sent? Currently while researching I only saw serviceuuid, etc.

Cierpliwy, 2017-12-12:
You can't access raw advertisement data on iOS.You are limited to:
https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerdelegate/advertisement_data_retrieval_keys

Jowin, 2017-12-12:
Thanks for the quick revert. Much appreciated.

closed 2017-12-12 by Jowin