dotintent/react-native-ble-plx #174: Mac install error

dllen, 2017-11-11:
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! react-native-ble-plx@0.6.4 postinstall: `node ./postinstall.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the react-native-ble-plx@0.6.4 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:

amrsekilly, 2017-11-11:
I have the same error too. I'm using it with create-react-native-app.

amrsekilly, 2017-11-11:
```
yarn add react-native-ble-plx@0.6.3
```
Worked for me.

petrbela, 2017-11-20:
From >=0.6.4 it should actually print the error message (in npm at least; yarn doesn't print it in the console):

```
Warning: Carthage is required to compile frameworks for iOS backend. You can install it with brew: brew install carthage. After installation go to ./node_modules/react-native-ble-plx and run ./build_ios_frameworks.sh or reinstall node module.
```

rknell, 2017-11-27:
Ok, so piecing all this together...

All of a sudden I couldn't build, tried yarn update and all sorts and nothing happened, dropped back to npm and got the same error above.

Tried a brew install carthage and it said it was already installed... but it could be updated!

TLDR; `brew update carthage`

Cierpliwy, 2017-11-30:
With new postinstall script I can see error message. Closing as it is not an issue of library itself. Hopefully with #29 we will be able to get rid of building issues once and for all.

closed 2017-11-30 by Cierpliwy

modularity, 2017-12-14:
Just installed the latest version of the library via npm and received the error "Failed at the react-native-ble-plx@0.7.0 postinstall script." I have successfully updated carthage with brew but still receive the error. Any suggestions on how to resolve this? For now, I installed 0.6.3 without error.