dotintent/react-native-ble-plx #176: Not able to get all manufactured data in Scan

Pradeep-DCPL, 2017-11-14:
Hi,

We are using this plugin to scan and connect to the BLE devices. We are facing one issue in reading the advertisement data in scanning that we supposed to get the 31 bytes data in Service data but getting only 9 bytes.

**BLE Device details- **  It is a custom-made hardware that is using nRF52832 SoC

**Below is the sample code that we are using** 

import { BleManager } from 'react-native-ble-plx';
this.manager = new BleManager();

//Checking Bluetooth state
const subscription = this.manager.onStateChange((state) => {
			if (state === 'PoweredOn') {
				this.startDeviceScan(scanTimeout);
				subscription.remove();
			}
	}, true);

//Start Scan
startDeviceScan(scanTimeout) {
		var me = this;
		this.manager.startDeviceScan(null, null, (error, device) => {
			if (error) {
				// Handle error (scanning will be stopped automatically)
				console.log("scanningError" + error.message);
				me.startDeviceScan();
				return;
			}
			if (device)
                            // Here we are supposed to get the advertisment data(Manufacturer data) in device object
				this.emit('startScanSuccess', device);
		});
	}

**Results we are getting in device object is as follows** 
serviceData - Object  =  {00006612-0000-1000-8000-00805f9b34fb:"/QosfgB5h/hZ↵"}
manufacturerData =  "P/8AAQ==" 

**Required result -** According to our device specification it should be 
Service data - We should get at least 31 Bytes of data.

Please help us to resolve this issue - may be we are missing some property or something else.
It is our app main functionality so please send your feedback as soon as possible.

**With the below plugin we are getting all the data.**
https://github.com/randdusing/cordova-plugin-bluetoothle
Thanks.

Cierpliwy, 2017-11-23:
Is this issue reproducible on Android/iOS or on both?

Pradeep-DCPL, 2017-11-23:
@Cierpliwy currently I have tested only on Android. Please let us know what can be the issue. 

Thanks, @Cierpliwy we need very urgent help on this so any input from your side will be very much appreciable.

Cierpliwy, 2018-01-05:
Any update on that one? Did you check advertisement data on iOS?

Cierpliwy, 2018-04-09:
Closing due to inactivity.

closed 2018-04-09 by Cierpliwy

shSina, 2019-07-02:
I have the same issue