dotintent/react-native-ble-plx #178: Secure BLE

fmaturel, 2017-11-22:
Hi, first thanks for all the good work done here.

This is more a question / feature request than an issue.
As far as I understand, there is no way today to get exclusive use of a BLE device. Whenever a React Native application can connect a device, any other app can connect it. Or another user using the same application could use my BLE device. How can this be secured?

If I use the device `monitorCharacteristicForDevice` method, how can I be sure that anyone else can do that too?

What would be the approach for securing a BLE device for an exclusive use?
Thanks for your thoughts.

Cierpliwy, 2017-11-23:
Starting from version 4.1 of BLE specification: "Slaves are permitted to have physical links to more than one master at a time.". Personally I worked with devices which allows only one active connection at a time and I guess this should be configurable on firmware side, but I'm not sure about it.

If you are interested in secure transmission of data, then it's another topic and I don't have any experience in that area.

fmaturel, 2017-11-23:
Thanks @Cierpliwy, I'll investigate both way of securing the connexion.
The drawback of allowing one connection at a time is that anyone can still connect to the device if the rightful user lost the connection, true?
Encryption of data seems to be a better approach.

Cierpliwy, 2018-01-05:
Issue #187 is related to BLE security. Closing this one.

closed 2018-01-05 by Cierpliwy