dotintent/react-native-ble-plx #181: [Feature request] Include base64 decoding capabilities

anderspitman, 2017-11-25:
First of all, thanks for the excellent library. I was impressed how quickly I got it up and running.

Unfortunately, I spent a sold hour or two hacking together a function to decode the base64 value from my device. My solution is also brittle and I think there's a good chance it will break on different data.

Would it be possible to integrate base64 decoding directly into the library? Is there some reason to actually want the data in base64 format?

watadarkstar, 2017-11-27:
There is a library called Buffer that can help with this.

**Decoding:**

```
import { Buffer } from 'buffer';
...
const buffer = new Buffer(characteristic.value, 'base64');
const bufStr = buffer.toString();
```

**Encoding:**

```
import { Buffer } from 'buffer';
...
const valueBase64 = new Buffer(jsonStr).toString('base64');
```

Cierpliwy, 2017-11-28:
>  Is there some reason to actually want the data in base64 format?

Due to limitation of JS <-> Native bridging we have to use json primitives to send data. Base64 is one of possible options to encode binary data as string and as it is very popular format I decided to support it. We could include base64 decoding library but:

* I don't want to force on user any particular implementation as they might want to use thier own
* I don't want to convert to binary format on JS side as data could be sent directly to REST API and additional overhead would be pointless.

Cierpliwy, 2017-11-30:
I will close this issue as we are blocked by RN to support sending raw bytes to JS side.

closed 2017-11-30 by Cierpliwy

anderspitman, 2017-11-30:
For my case at least, I realized that I was interpreting the encoding incorrectly, which was a big part of my problem. I also wasn't fully up-to-speed on handling typed arrays in Javascript. Brushing up on that helped a lot. Using base64 might not be the most intuitive but makes sense to me.