dotintent/react-native-ble-plx #182: yarn install fails when running from the root package

watadarkstar, 2017-11-28:
Running `yarn install` from the root of this package fails due to this line in `postinstall.js`.

https://github.com/Polidea/react-native-ble-plx/blob/f0d6eca6d391e624e943625ddd5adec5a078c2d2/postinstall.js#L4

It expects that a `package.json` exists 2 levels up, but that is not always the case if you are testing. i.e. I ran into this while I was trying to test my changes for #139.

dariuszseweryn, 2017-11-28:
Hello
If you test your changes locally- is it possible that you simply comment out this postinstall script check?

Cierpliwy, 2017-11-28:
During work on swift 4 support (swift4 branch) I added short patch which discards parent's config when building from non other destinations: https://github.com/Polidea/react-native-ble-plx/commit/4ad05ab525550ec00d26f04acdeba7c4a3b8173f.

watadarkstar, 2017-11-28:
@Cierpliwy Thank you! I put in a PR for this if you'd like to review and merge it: #183