dotintent/react-native-ble-plx #184: Communication only works with most recently connected device

rknell, 2017-11-29:
Hi,

I have a project where I need to connect two devices simultaneously. I connect both of them, one at a time, and only send additional write packets once the previous promise has resolved. I cache the device object returned from the scan and use it for communication.

The problem is the last device connected is the only one that responds with any data. Both of the devices write "successfully" but only the last connected performs any action and returns any data.

Does this library handle multiple connections? Happy to move this to stackoverflow if its more appropriate, but from what I can tell what I am doing should be working.

rknell, 2017-11-30:
Righto, so I found a bug!

When you call `<characteristic>.writeWithResponse()` internally it calls `writeCharacteristic`. In that function it searches for the characteristic UUID, but my devices are identical, and so it is only writing to one device.

As an immediate workaround, if I change from calling <Characteristic>.writeWithResponse(<data>) and instead call BluetoothManager.writeCharacteristicWithResponseForDevice, and specifying the device id, service UUID and characteristic UUID from the characteristic object it works.

I suggest we change `<characteristic>.writeWithResponse()` from calling `writeCharacteristic` and instead calling `writeCharacteristicForDevice`. I can't think of any regressions, but wanted to table it before I go about creating pull requests?

Cierpliwy, 2017-11-30:
It is not that easy.

Direct `writeWithResponse` was created to support multiple characteristics with the same UUID. By calling `writeCharacteristicForDevice` you are searching characteristic by UUID and first one matching your criteria will be returned.

I suspect that internal `characteristic.getId()` property returned by Android is not unique between devices but our implementation assumes that. It will need more work form my side I guess.

If your workaround works, just use it in mean time before I check and fix this.

Cierpliwy, 2018-01-05:
This should be fixed with version 0.8.0. Please reopen if you find any problems.

closed 2018-01-05 by Cierpliwy