dotintent/react-native-ble-plx #186: Scanning with Nexus 5x Bluetooth 4.2 failed

jchirschy, 2017-11-29:
Hi,

I'm trying to scan using a central `Nexus 5x` using this wonderful library, but in vain.
No peripheral is detected.

The same app is working fine with phone like OnePlus One or Honor 5x.
Using `Sniffator` or  `nRF Connect for Mobile` on this Nexus, the peripheral is detected and discovered successfully.

I 've set `LogLevel` to `Verbose`, but I got any help.

Is there something known related with Bluetooth 4.2 or with Nexus 5x ?
Is Sniffator using `react-native-ble-plx` or `RxAndroidBle` directly ?
How can I get some more debug to investigate this issue ?
I'm still using `0.6.3`. Is there any hope that something goes better with `0.6.4` ?
Any help would be greatly appreciated.

Thanks in advance.
JeanClaude

Cierpliwy, 2017-11-29:
You can always use latest version 0.6.5 which added functionality to negotiate MTU on Android. However there are no changes regarding scanning. I'll try to check tomorrow if we have Nexus 5x in Polidea and test it if that's the case. 

In mean time you can paste your Android logs on gist and link it here.

jchirschy, 2017-11-29:
Hi @Cierpliwy,

Unfortunately, I got no Android logs with LogLevel set to 'Verbose'.
When I do a logcat with `Sniffator`, I've got a lot...

I thank you in advance for your help.

Cierpliwy, 2017-11-29:
One more question are you filtering by service UUIDs?

jchirschy, 2017-11-29:
Yes, I'm filtering by service UUIDs

rknell, 2017-11-30:
Heads up I am using a 5x for testing my devices and its working for me, not sure if it helps but might point you in the right direction

jchirschy, 2017-11-30:
Hi @rknell,

What version of `react-native-ble-plx` do you use ?
Is there a possibility to get some codes with the Android manifest file to see if there could be some issues in my code which are highlighted only in a pure android stack ?

Cierpliwy, 2017-11-30:
I found Nexus 5X in Polidea (Android 8.0.0):

I tested 3 options:
`manager.startDeviceScan(null, null, ...)` - works as expected
`manager.startDeviceScan([""00e2f93e-3066-4e44-a1fc-83389d174d07""], null,...)` - works as expected (16bit UUID)
`manager.startDeviceScan(["8e0f"], null, ...)` - works as expected (128 bit UUID)

I didn't test for 16bit UUIDs as we don't have advertising devices with this type of UUID.

Before scanning by services make sure that your device really advertises them. They should be visible in `device.serviceUUIDs` property. If you can't find them here but you are sure they should be present, please send me your raw advertisement data:

- in react-native-ble-plx library go to android/src/main/java/com/polidea/reactnativeble/BleModule.java
- put breakpoint in line 278 (for version 0.6.5)
- send me value of `rxBleScanResult.getScanRecord()` of your device.

jchirschy, 2017-12-01:
Hello,

Finally, my Nexus 5 discovered successfully a test peripheral (`Nordic`).
It seems to be related to Android Manifest which was not good enough for a pure Android stack : Enabling location was not necessary with other centrals I was using before (my bad). 

Having modified Android Manifest, I was facing some weird caching issues apparently. Even an `adb kill-server` was not enough. Rebooting Mac and Nexus 5x, cleaning, rebuilding and enabling location was the right recipe.

I will try with the peripheral used in production next week. 
You're awesome guys with your support...

closed 2017-12-01 by jchirschy