dotintent/react-native-ble-plx #187: Implement and test pairing/bonding functionality

Cierpliwy, 2017-11-29:
Make sure that pairing and bonding functionality works properly on Android an iOS platform.

- [ ] Check what is currently working as expected
- [ ] Check which API surface is similar and different on both platforms
- [ ] Design common API
- [ ] Test implementation

tristandl, 2018-02-26:
does anyone have any idea about how complex this is to implement?

dariuszseweryn, 2018-02-26:
This topic should be relatively easy to implement. The catch is that (from my knowledge) iOS does not give any control over pairing/bonding to the user and Android is somewhat buggy in that matter (bonds being kicked off at apparently random moments).

tristandl, 2018-02-26:
that pretty much sums up what i thought already - mostly i'm concerned about fragmented implementations on the android side, and how many hoops there are to jump through to create a seamless js solution. iOS doesn't seem to be the problem, since it "just works" as they like to say. You can't un-bond I suppose, but you don't have the same android problems like "accessing a protected characteristic will kick off bonding, but not retry the read; or might just fail silently; or might fail explicitly; or .. or..." 

android!!! 😩

but seriously, i don't know the state of play - my gut feeling was that if the core library here decided not to do it, it was complex. if it's not that complex, point me in the right direction and i'm happy to give it a go (if it IS that complex, point me anyway 👍)

dariuszseweryn, 2018-02-27:
> accessing a protected characteristic will kick off bonding, but not retry the read; or might just fail silently; or might fail explicitly; or .. or...

You are probably referring to the `SweetBlue` big list of BLE issues. If so—I can just say that I have not encountered fragmentation this big. I have always seen the first scenario. But even then it should be fairly easy to cover most scenarios once the user would specify which operation can potentially kick-off bonding (as doing it for every connection could be a bit of overdo)

The core library does not have any helpers in regards of bonding because there were no need for that. Secondly there is little access to peripherals that are proven to handle bonding correctly (so all issues with bonding would be confirmed to be caused by the OS or the implementation)

State of the play? First it could be useful read these stackoverflow discussions [1](https://stackoverflow.com/questions/44472500/how-bluetooth-low-energy-security-works-between-android-app-and-ble-devices/44474281#44474281) and [2](https://stackoverflow.com/questions/18197444/android-bluetooth-low-energy-pairing/20093695#20093695) then probably to get into the official [bonding issues](https://issuetracker.google.com/issues?q=android%20bonding%20pairing)

tristandl, 2018-02-27:
awesome thanks for the tips!

dariuszseweryn, 2018-02-27:
Glad to help a bit. Feel free to report back with your findings! :)

assafb81, 2019-01-09:
any updates regarding bonding in Android and iOS?

closed 2019-03-23 by stale[bot]

alexzobi, 2020-03-31:
I'd be curious to know if anyone has figured out the pairing with security. I know people have said that with ios, sending a read/write command to a characteristic should cause the prompt for the code input, but I don't get that. i can see the device and get the arrays of services/characteristics/descriptors, but as soon as i try to read/write, i just get errors. no prompt for authentication code. the other weird thing is that the iphone doesn't pick it up on the bluetooth list when i try to bond through my settings. so why i can see it and semi-interact with it through this library, but iphone won't "discover" it in bluetooth settings is odd to me. any have any ideas?

dom96, 2020-12-08:
I'm seeing similar behaviour on Android, when I access a characteristic I see no prompt. Just a "Operation was rejected" error. Maybe I need to get my Android device to forget about the pair or something? But I think I will just set up a custom pairing process for my device.