dotintent/react-native-ble-plx #193: Send data as byteArray instead

dominikeinkemmer, 2017-12-06:
Hi,

I was thinking about using this module for one of my applications. However, I need to send data as a byte array, so I wondered if this is possible, as the documentation only states base64 data?

The company I'm working together with forces me to use a byteArray for the data I'm sending them so I have no way around it.

Thanks in advance.

Cierpliwy, 2017-12-06:
This is explained in detail in #181. In summary: just use base64 encoder and decoder.

pstanton, 2017-12-07:
Przemek, any idea why the base64 encoded string has line breaks in it and at the end?

Cierpliwy, 2017-12-07:
It is a bug which was introduced by my mistake...

On iOS I can set encoding/decoding options to:
```swift
// Use zero or one of the following to control the maximum line length after which a line ending is inserted. No line endings are inserted by default.
public static var lineLength64Characters: NSData.Base64EncodingOptions { get }
public static var lineLength76Characters: NSData.Base64EncodingOptions { get }

// Use zero or more of the following to specify which kind of line ending is inserted. The default line ending is CR LF.
public static var endLineWithCarriageReturn: NSData.Base64EncodingOptions { get }
public static var endLineWithLineFeed: NSData.Base64EncodingOptions { get }
```

Currently we are using: `self.base64EncodedString(options: .endLineWithCarriageReturn)`

On Android:
```java
/**
* Default values for encoder/decoder flags.
*/
public static final int DEFAULT = 0;
/**
* Encoder flag bit to omit the padding '=' characters at the end
* of the output (if any).
*/
public static final int NO_PADDING = 1;
/**
* Encoder flag bit to omit all line terminators (i.e., the output
* will be on one long line).
*/
public static final int NO_WRAP = 2;
/**
* Encoder flag bit to indicate lines should be terminated with a
* CRLF pair instead of just an LF.  Has no effect if {@code
* NO_WRAP} is specified as well.
*/
public static final int CRLF = 4;
/**
* Encoder/decoder flag bit to indicate using the "URL and
* filename safe" variant of Base64 (see RFC 3548 section 4) where
* {@code -} and {@code _} are used in place of {@code +} and
* {@code /}.
*/
public static final int URL_SAFE = 8;
/**
* Flag to pass to {@link Base64OutputStream} to indicate that it
* should not close the output stream it is wrapping when it
* itself is closed.
*/
public static final int NO_CLOSE = 16;
```
Currently we are using `DEFAULT` (by mistake) previously was `NO_WRAP`
Let me test it and fix it. I will send PR.

pstanton, 2017-12-07:
ah - thanks!

in the mean time, workaround (on receive notification / read characteristic):

```
		let value = characteristic.value;
		value = value.replace(new RegExp(String.fromCharCode(10), "g"), "");
		decoded = Base64.toByteArray(value);
```

Cierpliwy, 2017-12-07:
Added PR: #194

dominikeinkemmer, 2017-12-12:
Sorry for the late answer. Thank you for the input, everything is working fine now :)

durgesh0001, 2018-01-01:
Hi i want to read data from Bluetooth..do you have any example?

Cierpliwy, 2018-01-05:
Issues with new lines inserted into Base64 values should be resolved with version 0.8.0. Please reopen if you find any problems.

closed 2018-01-05 by Cierpliwy

kemicofa, 2019-03-14:
So I've spent countless hours trying to understand why the commands I was sending to my external device using bluetooth was not working.

The device accepts byte arrays such as `[0x10, 0x01]`. I've tried so many different ways of encoding it in base64 with nothing actually working.

After working along side with another colleague we came to the conclusion that it just doesn't work because of how the ascii table works.

It's a great project, however I took an hour to switch libraries that accepts the ability to send byte arrays and it now works.

Please consider accepting byte arrays.

kemicofa, 2019-03-14:
I've been using version 1.0.2 and after spending countless hours trying to send commands to my device, it just doesn't work.

I tried to encode `[0x10, 0x01]` into base64, I've tried (I think) almost every possible way. I worked with a colleague yesterday to try and finally get it working. We came to the conclusion that it was an issue with the ascii table.

I switched libraries that accepts byte arrays and it works fine now.

AdamGold, 2019-05-05:
Someone managed to send byte array encoded as base64 using this library, without changing the peripheral protocol?