dotintent/react-native-ble-plx #199: BleClientManager framework not found for architecture x86_64 after running carthage

LingboTang, 2017-12-13:
I followed the exact steps in README, and when I tried to build my project with ble-plx, I got this error:

`ld: framework not found BleClientManager for architecture x86_64`

And here is my setting:

[image]

Then I tried manually add frameworks to my project like this:

[image]

But I still got this:

`ld: framework not found BleClientManager`

Is there anything I'm missing?

LingboTang, 2017-12-13:
My fault, 

I set the framework search path to header search path, close this.

closed 2017-12-13 by LingboTang

swaroopa94, 2018-03-07:
@LingboTang  Hi I tried everything I added framework seach path also but the app is getting crashed and the report is as follows:

`dyld: Library not loaded: @rpath/libswiftCore.dylib
  Referenced from: /Users/ss-macmini-133/Library/Developer/CoreSimulator/Devices/1E7C8CEA-3BB8-4312-AD76-BE10A3093D13/data/Containers/Bundle/Application/07776147-E23B-4ECB-B2E5-C095111AD428/Sniffator.app/Frameworks/BleClientManager.framework/BleClientManager
  Reason: image not found
(lldb)`

Can you tell me how to solve this

LingboTang, 2018-04-04:
@swaroopa94 Why are you using the Sniffator.app, it looks weird to me.