dotintent/react-native-ble-plx #201: Windows version?

wind-rider, 2018-01-02:
It would be great if this could be used on [React Native for Windows](https://github.com/Microsoft/react-native-windows)! Do you have any plans to add that to your roadmap?

Cierpliwy, 2018-01-03:
I don't have Windows machine, so I'm looking forward to contributions!

closed 2019-03-23 by stale[bot]

cinderblock, 2021-01-12:
I see references to people using this package with Windows in some of the issues. Has Windows support been added? I'm having a hard time finding a full-ish list of supported systems.

msridar, 2024-04-05:
Any update on this feature ?

noshaid-ali, 2025-02-27:
Any good news about to add support for Windows desktop app?

noshaid-ali, 2025-03-06:
@msridar @cinderblock @hansmbakker How did you guys connect BLE devices with react native windows desktop app? I need help

msridar, 2025-03-06:
No, I havent used this module

noshaid-ali, 2025-03-06:
So how you did BLE communication in react native windows desktop?

hansmbakker, 2025-03-06:
I have not used React Native in the end

cinderblock, 2025-03-07:
@noshaid-ali per my previous comment, it seems this package doesn't support Windows.