dotintent/react-native-ble-plx #202: Is there a way to cancel monitor of characteristic?

pstanton, 2018-01-03:
I'm adding a listener via monitorCharacteristicForService:

```
deviceC.monitorCharacteristicForService(Params.svcUUID, Params.ctlChr, (error, characteristic) =>
	this.characteristicNotification(error, characteristic)
);
```

however when I disconnect from a device via `device.disconnect();` my listener is called with an error "Error: Unknown error: BleDisconnectedException{....."

I would like to remove/cancel my monitor/listener prior to disconnecting to avoid this error, however cannot see in the API how this is possible.

I also think it would be a good idea to allow people to stop monitoring a characteristic without disconnecting.

Any suggestions?

elliottheis, 2018-01-03:
@pstanton it looks like you've forgotten to pass an arbitrary transation id to `monitorCharacteristicForService()`

i.e.
```
deviceC.monitorCharacteristicForService(
  Params.svcUUID,
  Params.ctlChr,
  (error, characteristic) =>
    this.characteristicNotification(error, characteristic),
  'someTransactionId',
);
```

once specified you should then be able to stop monitoring the characteristic by calling `bleManager.cancelTransaction('someTransactionId')`

hope that helps

Cierpliwy, 2018-01-03:
`monitorCharacteristicForService` also returns subscription with `remove()` function so setting transactionId is not necessary.

Cierpliwy, 2018-01-05:
Closing as this shouldn't be an issue. There are two methods to cancel monitoring as described above.

closed 2018-01-05 by Cierpliwy

StephaneJuban, 2019-03-23:
I'm using v1.0.3 but for me, when I call `remove()` on the object returned by `monitorCharacteristicForService`, it doesn't stop the listener callback.

Any idea why ?

StephaneJuban, 2019-03-23:
@Cierpliwy :)