dotintent/react-native-ble-plx #207: ManufacturerData in advertisement data is being cached even after calling `bleManager.destroy()`

joshreeves, 2018-01-05:
I am successfully able to parse data from the `manufacturerData` property of the `device` object during peripheral scanning. I then connect to the device successfully and send a `writeCharacteristic` command to a characteristic which changes the `manufacturerData` to a new value. I can then disconnect and use other BLE tools to connect and verify that the `manufacturerData` was changed successfully. However, when I scan again with my react-native app using `react-native-ble-plx`, the scan response shows the old, original `manufacturerData` value. If I keep disconnecting and reconnecting, then sometimes it will flush this cached `manufacturerData` value and show me the new value, but often even that does not work.

I've also tried completely destroying the `bleManager` instance between connections and that does not help. It seems that the `manufacturerData` is being cached at some lower level.

Any help is greatly appreciated!

Cierpliwy, 2018-01-22:
Does it happen on iOS or Android? Did you enable `allowDuplicates` if you are working on iOS?

Cierpliwy, 2018-04-09:
We have tested this functionality in our internal project and manufacturerData changes during scanning process. Closing for now.

closed 2018-04-09 by Cierpliwy