dotintent/react-native-ble-plx #209: Need to have a guidance for Thermal Printer Execution

buraky84, 2018-01-11:
Hi all;

I have been developing with react native for 1.5 years. Currently implementing a bluetooth connected thermal printer application.

I have successfully connected to my Printer by reading documentation and the issues section. I don't have experience about bluetooth programming in the past. Currently i got stuck about how to continue next for exm. how to send data to printer to print.

After the printer connection is successfull, i am running `servicesForDevice` & `characteristicsForDevice` methods as suggested from what i read. Then i tried to run some write commands for exm. `writeWithoutResponse` where i don't exactly know what it is doing. (i assumed that it would trigger something on my printer.)

I really need some guidance how to continue next.
 * Will be appreciated for any comments, guidance and / or documentation.
 * After succesfull connection to a bluetooth device how do we send commands to that device (via which method)? Should i learn all the AT commands about my printer and send it that way? (Maybe there are some helper libraries for that which i couldn't discover yet.)

Any help will be appreciated.
Cheers.
Burak

`connect(myDevice) {
        console.log("Connecting to device :");
        console.log(myDevice);
        this.manager.connectToDevice(myDevice.id)
            .then((device) => {
                return device.discoverAllServicesAndCharacteristics();
            })
            .then((device) => {
                this.manager.servicesForDevice(device.id)
                    .then((servicesForDevice) => {
                        console.log("servicesForDevice");
                        console.log(servicesForDevice);
                        return(servicesForDevice);
                    })
                    .then((servicesForDevice) => {
                        for (let s of servicesForDevice) {
                            this.manager.characteristicsForDevice(s.deviceID, s.uuid)
                                .then((characteristics) => {
                                    console.log("characteristics");
                                    console.log(characteristics[0]);

                                    if(characteristics[0].isWritableWithoutResponse) {
                                        characteristics[0].writeWithoutResponse("1B 40");
                                        characteristics[0].writeWithoutResponse("0A");
                                        characteristics[0].writeWithoutResponse("0A");
                                    }
                                })
                        }
            }).catch((error) => {
                console.log(error);
            });

        }).catch((error) => {
            console.log("connect error: " + error);
        });
    }`

dariuszseweryn, 2018-01-12:
Hello,

Public functions of the library are documented in the [`.js` files](https://github.com/Polidea/react-native-ble-plx/blob/master/src/BleManager.js#L645).

If you need to learn more about (without) response then below are the other names that can be searchable.
`Write without response` == `write command`.
`Write with response` == `write request`.
I do not know the specification of your printer so I cannot help much with that.
All writes return a `Promise` so it is best to always check how they will resolve.

Remember that all values sent to and retrieved from a characteristic need to be `Base64` encoded.

buraky84, 2018-01-12:
Hello @dariuszseweryn 

Thanks for your answer.
Yesterday, i succeeded to print raw text by random tries and below command worked for me. Your reference is showing the same so seems i am on the right way.

`this.characteristic.writeWithoutResponse(Buffer.from(command, 'hex').toString('base64'))`

Its a thermal printer btw working with ESC POS commands.

Now i am trying hard to print images, barcodes, qr codes but as i could see there is no library i could take advantage of for react native. Any suggestions would be great on this! Thanks again.

dariuszseweryn, 2018-01-12:
Unfortunately I do not know any libraries that could help you with that.
Anyway it does not seem to be a problem with the `react-native-ble-plx` so I am closing this issue.

Best Regards

closed 2018-01-12 by dariuszseweryn

buraky84, 2018-01-12:
ok cheers.

shubhamdeol, 2019-03-25:
@buraky84   checkout this library [https://github.com/januslo/react-native-bluetooth-escpos-printer](url)

PhamMinhKha, 2019-09-11:
@buraky84 hi gay. can you tell how to print image. I am very grateful if you help me

FazilMuhammed, 2020-10-12:
@PhamMinhKha @buraky84 @dariuszseweryn Hello guys..anyone got the solution for how to print images, qr codes or barcodes please update your ideas.how to print the images what are the steps.

innalytics34, 2024-11-23:
> Hello @dariuszseweryn
>
> Thanks for your answer. Yesterday, i succeeded to print raw text by random tries and below command worked for me. Your reference is showing the same so seems i am on the right [...]

I also facing same issue, any solution for print qrcode in react-native-ble-plx library.