dotintent/react-native-ble-plx #210: Expokit project iOS :module appregistry is not a registered callable module

honbaa, 2018-01-12:
n our project, we add the “react-native-ble-plx” package, and config the Xcode as described in https://github.com/Polidea/react-native-ble-plx but after finishing the configurations, we build it and run on the simulator (iOS 11.2), it failed.
[image]

the environment as follows:

expo ski version :23
macos version:10.13.2
xcode version: 9.2
simulator version: iPhone 8 iOS 11.2

Has anyone solved this problem?

dariuszseweryn, 2018-01-12:
Hello,
I have seen this problem recently but with a different module. I was doing an update of Expo SDK on a detached application and I forgot to update `EXSDKVersions.plist`. Maybe it is the case in your situation?

honbaa, 2018-01-15:
@dariuszseweryn thanks for your reply. We have made other choice to solve it. Thank you all the same.

closed 2018-01-15 by honbaa

honbaa, 2018-01-15:
@dariuszseweryn We have decided to use the pure react-native project rather than Expo project.