dotintent/react-native-ble-plx #212: The connection has timed out unexpectedly

dominikeinkemmer, 2018-01-13:
Hi,

first of all thanks for the great library, it works perfectly so far :)

However, I found an issue (not sure if I missed something in the docs that I have to do or if this is a bug). If I'm connected to my bluetooth device and then just turn the bluetooth device off (cut the power supply of it) my application presents me with the following screen:

[image]

I'm subscribing to all events that I could find in the docs and if the device disconnects then I remove all subscription (the latest, some are removed earlier like onStateChange). Am I missing to catch something important or is this a bug of the library?

I also had a log in my XCode (something from RCTEventEmitter about not subscriping to a certain event) but not sure if this is related or not, as I am subscribing to all state changes that the library provides according to the docs.

PS: I'm using version 0.8.0

Thanks in advance.

Cierpliwy, 2018-01-22:
Currently all operations which are dependent on connection state of a device will receive connection errors as well. I'm not sure if there is any predefined order which promises will receive this event first and which as the last ones.

I'm working on better Error signature, which will include generic error code and allow to handle it better in the future.

dominikeinkemmer, 2018-01-25:
I just was able to test on an android device again and a device that previously had no issue is now unable to find any bluetooth devices (the code for scanning didn't change though) can that have any relation to upgrading from `0.7.x` to `0.8` ? The affected devices seem pretty random to me (Samsung Galaxy Note 10.1 Tablet Android 4.3, Samsung SM-G361F with Android 5.3 and a huawai device Android 6.0.0)

Do you happen to know anything in relation to this? I can't debug on their phones as the testing is done remote between japan and Austria.

closed 2018-04-08 by Cierpliwy