dotintent/react-native-ble-plx #213: How to get response from the printer

honbaa, 2018-01-15:
Currently, we can successfully send commands to printer by the method "writeCharacteristicWithResponseForDevice", for example: feedpaper, handshake, and readRibbon.

But we want to get the response from the printer! For example, we want to know the name of the printer, the printer will return printer name when we send "handshake" command.  When we send "handshake" , the method "writeCharacteristicWithResponseForDevice" return a Promise<Characteristic>, however we don't get the right value by "Characteristic.value". The content of "Characteristic.value" is still "handshake" command.

So, is there a method for receiving the data that printer returns?

Cierpliwy, 2018-01-19:
Technically writing a value doesn't mean that it will be accepted by peripheral device. This is an iOS behaviour. In case you want to check value after your command, you should call read operation.

honbaa, 2018-01-22:
Thanks for your reply! I have solved this problem by the method “monitorCharacteristicForDevice”. And the characteristic is a notifiable one. Although the characteristic to be written is not the same one. But i can receive the notification from the notifiable characteristic!

closed 2018-01-22 by honbaa