dotintent/react-native-ble-plx #214: BleModuleInterface flow errors

stinodes, 2018-01-18:
React-native improved `NativeModule`-typing, causing the following flow-error:

```
Error: node_modules/react-native-ble-plx/src/BleManager.js:52
 52:     this._eventEmitter = new EventEmitter(BleModule)
                                               ^^^^^^^^^ BleModuleInterface. This type is incompatible with the expected param type of
 34:   constructor(nativeModule: ?NativeModule) {
                                  ^^^^^^^^^^^^ object type. See: node_modules/react-native/Libraries/EventEmitter/NativeEventEmitter.js:34
```

I have a fix ready, if you guys are open to pull-requests

Cierpliwy, 2018-01-19:
Thanks again!

closed 2018-01-19 by Cierpliwy

stinodes, 2018-01-19:
No problem, happy to contribute where I can!