dotintent/react-native-ble-plx #216: I want to get explicit mac address of devices on iOS

LingboTang, 2018-01-19:
I want to get explicit mac address of devices on iOS when I scan nearby devices on my phone and store the paired devices locally. But based on the discussion in https://github.com/Polidea/react-native-ble-plx/issues/92, the `Device.id` on iOS is 

> on iOS it will be autogenerated UUID

I want to get the explicit mac address somehow, is it possible for your team to add this attributes to iOS?

dariuszseweryn, 2018-01-19:
Hello @LingboTang 

Thanks for using the library!

Unfortunately iOS does not provide any access to the peripheral's MAC address on the native level—so no—it is not possible to add this attribute to iOS.

Best Regards

closed 2018-01-19 by dariuszseweryn

mateusz-, 2018-02-16:
One possibility is to add a [Device Information](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.device_information.xml) service with a [System ID](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.system_id.xml) characteristic.

react-native-ble-plx could probably read it "behind the scenes" once connected but it would be an "auto-magic" feature - probably not desirable.

1459726992, 2019-03-01:
I'm going to get my bluetooth MAC to verify to the server that it's legitimate and I don't know what to do
I sorry for my English

serdargun, 2020-03-23:
@1459726992 I have same problem. I need to access MAC address of peripheral devices to verify with the server. Did you solve it?

EduardoRicardez, 2020-05-21:
Did you solve it?

GuidoGagliardini, 2023-11-28:
Hello!
Has the error been fixed? or is there a workaround?