dotintent/react-native-ble-plx #217: Run as Android Foreground Service

rgstephens, 2018-01-20:
I've read some of the discussions about background mode (#127 & #151).  I have a basic RN project that starts an android Foreground Service and runs BluetoothLeScanner to scan for devices.  I only need to capture advertising packets (no connections, no GATT).

Can I use this project from my RN bridge foreground service?

Cierpliwy, 2018-01-22:
I didn't do it myself, but I can't see any potential blockers. Do you agree @dariuszseweryn , @paweljaneczek ?

dariuszseweryn, 2018-01-22:
I do not see any problems from the Android side apart from the UX.

rgstephens, 2018-01-23:
This is my first RN bridge and I'm not a native developer.  It seems like the best approach would be for my Java foreground service call the `com.polidea.reactnativeblue.BleModule` calls (startDeviceScan, onStateChange, stopDeviceScan, etc).

Does that make sense?

rgstephens, 2018-01-26:
I'd prefer to call it from JS but I have to start the foreground service from Java (don't I?) and wouldn't I have to make the startDeviceScan from within the Java foreground service code?

rgstephens, 2018-02-02:
If I start an Android Foreground Service, then on the JS side, periodically call `startDeviceScan`, the JS code is only executed while the app is active even though the Foreground Service is active and shown in notifications. If the screen times out or the user switches away from my app, the JS code stops running even though the Foreground Service is active.

dariuszseweryn, 2018-02-02:
@rgstephens You are not calling `startDeviceScan` from any `Component` do you?

rgstephens, 2018-02-06:
Yes, I was.  Good question.  I'm going to make a sample app and do the call the from the root.  This [example in github](https://github.com/rgstephens/ForegroundTest) currently starts a loop in `index.js` that logs the time to console.log and stores it in redux every 10 seconds. I start the loop before the root component is registered.

I have a switch on the main component that starts a foreground service.  Whether the service is started or not, the 10 second timer updates stop when the app loses focus, even though the Android Foreground Service notification shows that it is still running.

I'm going to update this loop to call startDeviceScan at a regular interval.

```
setInterval(function() {
    console.log('*** index.js setInterval', new Date());
    store.dispatch({ type: 'TIMESTAMP' });
  }, 10000);
```

rgstephens, 2018-02-08:
@DariuszAniszewski I now have a complete example on [github](https://github.com/rgstephens/ForegroundTest) which calls `startDeviceScan` from the root in a loop every 10 seconds and has a switch to start/stop a foreground service.  The scans still stop when I switch away from the app even though the foreground service is running.

DariuszAniszewski, 2018-02-08:
cc: @dariuszseweryn ^^

Cierpliwy, 2018-04-08:
@rgstephens Do you still have an issue?

rgstephens, 2018-04-08:
Yes, I still have the problem. The sample app posted announce does not work in the background.

closed 2019-03-23 by stale[bot]

Deityhub, 2019-06-20:
still facing this same issue too

bimix, 2020-01-24:
timers and intervals do not work in background mode. Use something else.