dotintent/react-native-ble-plx #219: Finding only one Device after Scan

vinipachecov, 2018-01-24:
Hi there,

No matter how mobile devices there are around with bluetooth on , I only find one after a scan. I've created an app with Kotlin and did the search and it could find all devices available as It was supposed to. Any help is welcome.

Mobile device tested  Moto E
OS: Linux Deepin 15.5
React Project: Native Code
Code of the project: https://github.com/vinipachecov/react-native-bluetooth

Here is the main part of the code:
[image]

As you can see it is a very basic react class code just with the functions of the library to scan devices and add to a state and later try to show it.
[image]

My render method:
[image]

piotrdubiel, 2018-01-24:
Hi!
The problem is in the way you're constructing `deviceList` and updating the state. React and React Native require you to use `this.setState` to trigger state update and later re-render as a result. If you use mutating function like `Array.prototype.unshift()` then your array's content will change, but it will not trigger component re-render.

Consider this example
```js

class Test extends Component {
  constructor() {
    super()
    this.state = { list: [] }
  }

  componentWillMount() {
    setInterval(() => {
      // this.state.list.unshift(new Date().getTime()) // No render :(
        this.setState(prevState => ({ // using this.setState will trigger render
          list: [new Date().getTime(), ...prevState.list],
        }))
    }, 1000)
  }

  render() {
    console.log("render")
    return (
      <View>
        <Button onPress={() => this.forceUpdate()} title="Force update"/>
        <Text>The list of length {this.state.list.length}:</Text>
        {this.state.list.map((v, i) => <Text key={i}>{v}</Text>)}
      </View>
    )
  }
}
```
It's pretty similar to what you have now, because it asynchronously "updates" state each second. The problem is, if it calls unshift then only forcing update will show the content of the list.
On the other hand, if the `setState` is used then updates each second with new timestamp.
Generally, the better practice is to follow immutable approach and always returning new object, like in this example with spread operator.

I hope it clarifies things :)

closed 2018-01-25 by dariuszseweryn

vinipachecov, 2018-01-25:
Thank you @piotrdubiel. The issue of finding only one value has been solved. Still I couldn't find my Iphone even after refactoring the code.

dariuszseweryn, 2018-01-25:
Does your iPhone advertise itself?