dotintent/react-native-ble-plx #220: Support for peripheral mode.

Cierpliwy, 2018-01-26:
Currently library does not support peripheral mode, which will allow to send data via BLE between phones. This issue is blocked by support in it's dependent libraries:

* RxBluetoothKit (Polidea/RxBluetoothKit#34)
* RxAndroidBle  (Polidea/RxAndroidBle#152)

EskelCz, 2018-10-16:
@Cierpliwy I could really use this, how can I help?

1. Seems like RxBluetoothKit already has peripheral mode implemented. Correct?
2. That required a java programmer? Any idea how much work would it take?
Maybe I can open bounty on bountysource that would fund it.

EskelCz, 2018-10-23:
@Cierpliwy Supposedly the second issue should not be a show stopper and vanilla android ble server can be used: https://github.com/Polidea/RxAndroidBle/issues/152#issuecomment-432244335

EskelCz, 2018-12-23:
Also here is the android implementation done, I can imagine the author won't object if it's used in this library. I can ask him, we're in contact about the ios counterpart.
https://github.com/himelbrand/react-native-ble-peripheral

EskelCz, 2019-01-29:
@Cierpliwy I'm just finishing ios version for the react-native-ble-peripheral library (see above)
polidea doesn't seem interested

closed 2019-04-06 by stale[bot]

GuillaumeRemyCSI, 2020-04-29:
Any update on this ? I also need this functionality for detecting phones at proximity

WhiteWolph, 2020-05-31:
looking for a collaboration, or possibility of a library to have peripheral and central functionality combined. i have a feeling there is one already, since ive been curious how the 'Covid tracer api' made by google and apple works?  i guess they already have that ?

Thanks in advance.

EskelCz, 2020-05-31:
> since ive been curious how the 'Covid tracer api' made by google and apple works? i guess they already have that ?
Yeah, me too. I had a project that needed something similar to the contact tracing and it was nearly impossible to do. Too draining for the battery to advertise and scan all the time. If you find something please let me know. I invested a ton of time into it and got nowhere. :(