dotintent/react-native-ble-plx #221: `React/RCTBridgeModule.h` file not found when following set up instructions using ExpoKit

tsemerad, 2018-01-30:
I created a brand new project using `create-react-native-app` then ejected it (keeping ExpoKit), confirmed that it runs fine from XCode, then followed the [Configuration and Installation steps](https://github.com/Polidea/react-native-ble-plx#configuration--installation). Then when I try building from XCode it tells me `'React/RCTBridgeModule.h' file not found`
[image]

A coworker of mine also ran into the same error when following the setup instructions in the documentation. Is this a known issue when working with a project detached from Expo?

I'm using `expo@25.0.0` and `react-native-ble-plx@0.8.0`.

Cierpliwy, 2018-01-30:
This library was tested only with projects created by command `react-native init`. ExpoKit have different build process, which is painful to support right now, when you are dependent on dynamic Swift libraries.

Cierpliwy, 2018-01-30:
@dariuszseweryn Has more experience with it than me.

dariuszseweryn, 2018-01-30:
The problem is that `react-native-ble-plx` has no information about other dependencies which are managed in `Pods` when using `Expo`. The easiest fix is to add `$(SRCROOT)/../../../ios/Pods/Headers/Public/**` to the `BleClient` module in `XCode`.

To automate the fix one can create a git diff/patch with the above change and apply it as a `postinstall` script

Best Regards

tsemerad, 2018-01-30:
Thanks for the response, I really appreciate it. Adding `$(SRCROOT)/../../../ios/Pods/Headers/Public/**` to the Header Search Paths for `BleClient` got me past the error, but now I'm getting a different error:
`clang: error: no such file or directory: '/Users/tanner/Projects/ticketing-client/node_modules/react-native/ReactCommon/yoga/yoga/YGEnums.c'
clang: error: no input files`

Searching on that takes me to [this issue](https://github.com/facebook/react-native/issues/11781) which implies it's related to swift usage. Do you have an idea of how to work around this new issue?

dariuszseweryn, 2018-01-30:
Unfortunately I have no idea where this next error comes from but it doesn't seem related to `react-native-ble-plx`. Correct me if I am wrong

JofArnold, 2018-02-23:
I'm not an iOS dev so I'm not at all sure I'm doing the right thing, but I found I could fix it by going through all the errors and making the following changes.

```
#import <React/RCTBridgeModule.h>
```

With

```
#import "RCTBridgeModule.h
```

etc. Not sure if it's related but I'm using the Expo fork of React Native

dariuszseweryn, 2018-02-24:
Closing as the usage with Expo is already described [here](https://github.com/Polidea/react-native-ble-plx#expo)

closed 2018-02-24 by dariuszseweryn