dotintent/react-native-ble-plx #222: Scan throttle exception

andycamp, 2018-02-01:
I get this if I lose connection and call connection from the edge of coverage:

scan error { message: 'Unknown scan error: BleScanException{reason=UNDOCUMENTED_SCAN_THROTTLE, retryDateSuggestion=Thu Feb 01 10:23:53 EST 2018}',
  code: 800 }

Do you have any advice for avoiding this exception, or handling it correctly in code when connection drops?

dariuszseweryn, 2018-02-01:
Hello

This exception is from [Android](https://github.com/Polidea/RxAndroidBle/blob/master/rxandroidble/src/main/java/com/polidea/rxandroidble/exceptions/BleScanException.java#L84). To avoid it one must not start more than 5 scans during a 30 seconds window.

andycamp, 2018-02-01:
Thanks. We were suspecting something like that, but wanted to try to understand it better. We put a timer in to wait a few seconds between attempts and it seems to have resolved the issue for us.

closed 2018-02-01 by andycamp