dotintent/react-native-ble-plx #225: A problem occurred configuring project ':react-native-ble-plx'.

ProMasoud, 2018-02-12:
I installed this package but at first return error while building app
I installed in fresh native-script too, but again same error

**Error:**
```

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':app'.
> Could not resolve all dependencies for configuration ':app:_debugApk'.
   > A problem occurred configuring project ':react-native-ble-plx'.
      > Could not resolve all dependencies for configuration ':react-native-ble-plx:classpath'.
         > Could not resolve com.android.tools.build:gradle:2.2.0.
           Required by:
               sniffator:react-native-ble-plx:unspecified
            > Could not resolve com.android.tools.build:gradle:2.2.0.
               > Could not get resource 'https://jcenter.bintray.com/com/android/tools/build/gradle/2.2.0/gradle-2.2.0.pom'.
                  > Could not GET 'https://jcenter.bintray.com/com/android/tools/build/gradle/2.2.0/gradle-2.2.0.pom'.
                     > Connect to jcenter.bintray.com:443 [jcenter.bintray.com/75.126.118.188] failed: Connection timed out: connect
         > Could not resolve org.codehaus.groovy:groovy-android-gradle-plugin:1.1.0.
           Required by:
               sniffator:react-native-ble-plx:unspecified
            > Could not resolve org.codehaus.groovy:groovy-android-gradle-plugin:1.1.0.
               > Could not get resource 'https://jcenter.bintray.com/org/codehaus/groovy/groovy-android-gradle-plugin/1.1.0/groovy-android-gradle-plugin-1.1.0.pom'.
                  > Could not GET 'https://jcenter.bintray.com/org/codehaus/groovy/groovy-android-gradle-plugin/1.1.0/groovy-android-gradle-plugin-1.1.0.pom'.
                     > Connect to jcenter.bintray.com:443 [jcenter.bintray.com/75.126.118.188] failed: Connection timed out: connect

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 1 mins 36.828 secs
Could not install the app on the device, read the error above for details.
Make sure you have an Android emulator running or a device connected and have
set up your Android development environment:
https://facebook.github.io/react-native/docs/android-setup.html

```

my config is:
```
"dependencies": {
    "react": "16.0.0",
    "react-native": "0.50.4",
    "react-native-ble-plx": "^0.7.0",
    "react-redux": "^5.0.6",
    "redux": "^3.7.2",
    "redux-thunk": "^2.2.0"
  },

node: v8.9.4
windows 10 and android device
```

Cierpliwy, 2018-02-15:
As you can see there are connection problems during fetching dependencies on Android:
`failed: Connection timed out: connect`
That's the reason of failure. Are you building directly from Android Studio or via `react-native run-android`?

Cierpliwy, 2018-04-08:
@ProMasoud Closing due to inactivity. Please reopen if issue still occurs.

closed 2018-04-08 by Cierpliwy

amerizade73, 2018-10-17:
i have sane issue :'(

amerizade73, 2018-10-17:
I bough a VPN and 443 Error Solved for me...