dotintent/react-native-ble-plx #227: onDisconnected called multiple times for a disconnected device

vldgeorgiev, 2018-02-15:
I noticed some behavior that seems strange with the onDisconnected listener. If a device is connected and then disconnected, the listener is called once, as it should. But if the same device connects and disconnects again in a short period, the listener is called twice. Then on the next disconnect three times and so on.
This is on Android with ble-plx 0.8.0.

My simplified code is something like this:
```
    bleManagerPlx.startDeviceScan([SERVICE], { allowDuplicates: false }, (error, device) => {
      if (device.name.startsWith('......')) {
        if ('a check that the device is not already connected') {
          connectDevice(device.id);
        }
      }
    })

  bleManagerPlx.connectToDevice(deviceId)
    .then((device) => {
      device.onDisconnected((error, disconnectedDevice) => {
        console.log('Disconnected ', disconnectedDevice.name);
      });
    });
```

Now, I could store the subscription that is returned by onDisconnected() and call remove on it to stop listening, but I think that the subscription should have been removed by the manager when the connection is cancelled.
Is that by design?

Thanks.

Cierpliwy, 2018-02-15:
It looks like a bug. I'll check it tomorrow and let you know what is the cause of the issue.

dariuszseweryn, 2018-04-26:
@vldgeorgiev 
If you use call this:
```
bleManagerPlx.connectToDevice(deviceId)
    .then((device) => {
      device.onDisconnected((error, disconnectedDevice) => {
        console.log('Disconnected ', disconnectedDevice.name);
      });
    });
```
For each connection then on each connection you register a new disconnection listener.

Do you think that each listener should be called only once? It is possible to unsubscribe a listener when it is no longer needed

vldgeorgiev, 2018-04-27:
@dariuszseweryn Well, yes, I register a new listener once a device is connected. I think it doesn't matter if it is a new device, or a reconnect of an old one after a minute, an hour or a day.
At least to me it seems better to automatically unsubscribe from the old listener once a device disconnects. This device may never reconnect, so why should the listener stay active.

TrebuhD, 2019-01-29:
Is there a way to check whether a device has an `onDisconnected` listener already attached? It's problematic to have to figure out a way to store the references to the `Subscription` objects.

dariuszseweryn, 2019-02-01:
@TrebuhD Unfortunately there is no way to check if any listener is attached

closed 2019-04-09 by stale[bot]

will-thrive, 2023-01-03:
I see the same issue using react-native@0.70.2 and react@18.1.0: first time the message appears twice, then thrice, then 4 times etc.