dotintent/react-native-ble-plx #228: Android studio emulator dont detect any device !

karimraoudy, 2018-02-15:
Hello, thank you for this great library, I work on a project and I need to connect the application with a specific device, but it is not detected in my emulator, and when I try the sniffer, it is detected and connect successfully. the library work with real device or it work with the android studio emulator?
thank you all.

dariuszseweryn, 2018-02-24:
Hello,
Thanks for using the library!
Unfortunately [no Bluetooth functionality works on Android Emulator unless some steps are taken](https://stackoverflow.com/questions/22604305/how-to-use-android-emulator-for-testing-bluetooth-application)
Best Regards

closed 2018-02-24 by dariuszseweryn

karimraoudy, 2018-02-26:
Ok, thank you  @dariuszseweryn