dotintent/react-native-ble-plx #229: Characteristic 0003cdd1-0000-1000-8000-00805f9b0131 notification change failed: The attribute could not be found.

honbaa, 2018-02-23:
I use "monitorCharacteristicForDevice" method to receive notification from device. but error occurs as follows:
'Characteristic 0003cdd1-0000-1000-8000-00805f9b0131 notification change failed: The attribute could not be found.'

I don't know why this happened. Does anyone know ?

honbaa, 2018-02-24:
I have found that this error discribed at https://github.com/Polidea/react-native-ble-plx/blob/master/ErrorMapping.md#error-mapping-for-react-native-ble-plx as follows:

characteristicNotifyChangeFailed(Characteristic, Error?) | **Notify change** | ???

I can't understand the reason really. Any ideas?

honbaa, 2018-02-24:
@corymsmith @iegik @literator

iegik, 2018-02-26:
> Does your characteristic has a `PROPERTY_NOTIFY`?
> I think that your characteristic (0003cdd1-0000-1000-8000-00805f9b0131) does not have a Client Characteristic Config descriptor

- answered by @dariuszseweryn in https://github.com/Polidea/RxAndroidBle/issues/202

Cierpliwy, 2018-04-08:
@honbaa Please reopen if needed.

closed 2018-04-08 by Cierpliwy

AntonioAlfrz, 2018-04-20:
Hello,

before opening a new issue I would like to ask first because I think I have a similar problem.
I am facing the error:
"Cannot find client characteristic config descriptor (code 2) with characteristic UUID 1ed9e2c0-266f-11e6-850b-0002a5d5c51B" (Error code  403) when trying to monitor a characteristic wiht the function monitorCharacteristicForService, although the characteristic is notifiable.

I have been able to subscribe to changes using other libraries.

Thank you.

dariuszseweryn, 2018-04-21:
@AntonioAlfrz I am guessing you have tested on Android and that you are trying to monitor characteristic changes on some device that violates Bluetooth Spec by not having a `Command Characteristic Configuration Descriptor` under a characteristic that has a `Notify` property?

AntonioAlfrz, 2018-04-21:
Yes, Android 6.0.1 and the device is  [bosch XDK](https://xdk.bosch-connectivity.com/)

dariuszseweryn, 2018-04-21:
I have just checked my old code for XDK interactions. It indeed does not have a proper CCC descriptor attached to notification characteristics. As for Today the newest version of the library `0.9.0` does not handle this case. You would need to revert to `0.8.0` which in turn does not correctly cancel monitoring of characteristics that do have CCC descriptor

yashkmochi, 2018-11-14:
It can happen , if your characteristic is not notifying or indicating.
Or,  your characteristic might not have a Client Characteristic Config descriptor which can cause the error.
If that is the case you could use function RxBleConnection.setupNotification(UUID, NotificationSetupMode.COMPAT) as a workaround. in this location ('../node_modules/react-native-ble-plx/android/src/main/java/com/polidea/reactnativeble/BleModule.java')

Andy4ward, 2018-11-30:
@yashkmochi how do you access that function from javascript?

I've been chasing around the package to see if it's possible but coming up empty

yashkmochi, 2018-12-14:
@Andy4ward 
I am not using that function direclty from javascript. It is internally being used while subscribing to specific characteristic using ble-plx.

asurare, 2019-04-25:
I had a issue pretty similar, but on iOS. I had to modify the swift sources in `react-native-ble-plx⁩/ios⁩/RxBluetoothKit/Peripheral.swift⁩` with the following changes in the `setNotifyValue` line 326, and commenting the `throw BluetoothError` : 

```
 let observable = peripheral
            .rx_didUpdateNotificationStateForCharacteristic
            .filter { $0.0 == characteristic.characteristic }
            .take(1)
            .map { (_, error) -> Characteristic in
                if let error = error {
                    // throw BluetoothError.characteristicNotifyChangeFailed(characteristic, error)
                }
                return characteristic
            }
```

packetstracer, 2019-05-24:
@asurare same issue happening to me, check the bug I opened if you want to get more info. 

[https://github.com/Polidea/react-native-ble-plx/issues/468](url)

What bluetooth peripheral are you trying to connect from iOS? I'm using an ESP32 (Arduino compatible) device.

asurare, 2019-05-24:
@packetstracer The url behind your link is broken ^^

PR #385 (2019-01-11): Fixed notifications/indications on characteristics w/o CCC descriptors
According to the specification of Bluetooth: Client Characteristic Configuration Descriptor is obligatory for characteristics that expose notification/indication property. Some devices are not complying to the specification. To make the RxAndroidBle library work in these situations a special setup mode (NotificationSetupMode.COMPAT) was introduced.

Fixes #229