dotintent/react-native-ble-plx #230: Turn off or refresh device caching on iOS

LingboTang, 2018-02-23:
I'm trying to scan some nrf52 UART devices, I have no problem scanning them. However, when I changed the device name, and then do a scan using ble-plx again, it still displaying the old name that I believe it is cached in iOS system. I confirmed this behaviour with nordic nrf connection app. What's different is that nordic app will refresh the device name after few seconds or after enable a connection between my iphone and my device. I assume ble-plx has similar functions to refresh the cache, but I didn't see it explicitly in the documentation. Can someone help?

Cierpliwy, 2018-04-08:
iOS uses value of https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.gap.device_name.xml in caches, so yes it is possible that when device name is changed and new connection wasn't established scanning can show old name.

Unfortunately I'm not aware of any specific API on iOS to make this process faster. We are using currenty https://developer.apple.com/documentation/corebluetooth/cbperipheral/1519029-name.

closed 2019-03-23 by stale[bot]

CyxouD, 2020-06-19:
Still not implemented, but I can only say for sure that it happens on Android

gilador, 2020-06-23:
i think they way to refresh the gatt cache is by passing the right option parameter:
`connectToDevice(deviceIdentifier: DeviceId, options?: ConnectionOptions): Promise<Device>`
where ConnectionOptions has `refreshGatt?: RefreshGattMoment`:
[image]

originalix, 2022-06-23:
> i think there is a way to refresh the gatt cache by passing the right option parameter to the connectToDevice method: `connectToDevice(deviceIdentifier: DeviceId, options?: ConnectionOptions): Promise<Device>` where ConnectionOptions has `refreshGatt?: RefreshGattMoment`: [image] so the call should be: `const result = await this.manager.connectToDevice(deviceId, { refreshGatt: 'OnConnected', ...other-options.... })`

`refreshGatt` option is Android only.

ArthurRuxton-DY, 2022-12-05:
@originalix @gilador @Cierpliwy @LingboTang 

> `refreshGatt` option is Android only.

Is there a way to refresh or clear cache on IOS?

nmurashi, 2023-06-14:
Is there any update to this for iOS? Right now it seems the only way to refresh the device in the cache is to connect to it.