dotintent/react-native-ble-plx #231: Can't find all nearby bluetooth devices

qlerebours, 2018-02-26:
Hello,
thanks for your work on this library !

I'd like to get more information about devices that can be scanned because there are devices that I can't scan.
When I go to the bluetooth parameters on my phone (Xiaomi MI5) and I search, i can find my headphone (Jabra MOVE v2.5.0). If I scan with the following code, I can find some devices, but not this one. 

```
scanAndConnect() {
        let deviceNames = [];
        this.manager.startDeviceScan(null, null, (error, device) => {
            if (error) {
                // Handle error (scanning will be stopped automatically)
                return
            }

            if (device.name) { // Creating an array of devices with names
                if (deviceNames.indexOf(device.name) === -1) {
                    deviceNames.push(device.name);
                    console.log('Found new device: ', device);
                }
            }

            if (device.name === 'Jabra MOVE v2.5.0') {
                console.log('Trying to connect to '.device.name);

                this.manager.stopDeviceScan();

                device.connect()
                    .then((device) => {
                        let charac = device.discoverAllServicesAndCharacteristics();
                        return charac;
                    })
                    .then((device) => {
                        console.log(device);
                        console.log('Do work on device with services and characteristics');
                    })
                    .catch((error) => {
                        console.log(error);
                    });
            }
        });
    }
```

Is there something i'm doing wrong ? If my phone find a device with a name, will it have a name when I run `scanAndConnect` ?

Thanks

Cierpliwy, 2018-03-13:
Name field is not very stable. On iOS after first successful connection operating system will try to read:
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.gap.device_name.xml
which may result in slightly different name then one found in advertisement data. It will cache this value even after disconnection.

AtlasMaxima, 2018-04-10:
I also been having the same problem. I'm using Galaxy Tab A6 which supports BLE and Scanner Series CS3070 Motorola. 
The Scanner CS3070 is discoverable by the tablet but when I try to scan with the below code, it can't not find the device at all. 
`export default class App extends Component {
  constructor(props) {
    super(props);
    this.state = {text: '', change: ''};
    this.manager = new BleManager();
  }

  searchBluetoothDevices = () => {
    this.state.deviceList = []
    console.log('Started searching devices');
    var i = 0;

    this.manager.startDeviceScan(null, null, (error, device) => {
      if (error){
        this.setState({actionResult: error.message});
        return
      }
      console.log( `Found Device ${device.id}. ${i} is connected = ${device.isConnectable ? null : 'Not connected'}
      Device name = ${device.localName ? null : 'No name for this Device'}`);
      this.state.deviceList.unshift(device);
      if(device.id == 'CS3070:12332522502688'){
        this.manager.stopDeviceScan();
      }
      i++;
    });
  }`

Cierpliwy, 2018-04-10:
What do you mean by `The Scanner CS3070 is discoverable by the tablet`? Are you talking about Bluetooth settings or some application like nRF Connect?

qlerebours, 2018-05-14:
@Cierpliwy before posting I tried to find my headphone specifications but I didn't find anything. I worked and learned many things on bluetooth and I think that indeed it was not supporting Low Energy.
To be sure that the library was working correctly, I ran a BLE server on a Raspberry to check if it was discoverable and it worked: I found the device and was able to connect it.
I think that this is the same problem for you @AtlasMaxima . Moreover, you have to know the Bluetooth Low Energy is not physical but logical. For example, a Raspberry can support either standard bluetooth and also BLE.

Mhde96, 2020-10-21:
i have the same probleam