dotintent/react-native-ble-plx #232: Image not found error for framework

swaroopa94, 2018-03-07:
Hi,

I installed the library and followed the all steps but when running appilication it is crashing wth following output in ios

dyld: Library not loaded: @rpath/libswiftCore.dylib Referenced from: /Users/ss-macmini-133/Library/Developer/CoreSimulator/Devices/FDF981CA-7FA5-4680-BDA7-14D2D9D88C10/data/Containers/Bundle/Application/490EAC9C-4B00-49D9-B575-0F39C234A893/reduxsample.app/Frameworks/RxBluetoothKit.framework/RxBluetoothKit Reason: image not found (lldb)

Can you tell me how to solve htis

syam82, 2018-03-13:
Hello why there is no response please reply as soon as possible

Cierpliwy, 2018-03-13:
I guess it is related to following step:
8) In `Build Settings`/`Build Options`/`Always Embed Swift Standard Libraries` set to `Yes`.

bacancy-swaroopa, 2018-03-13:
@Cierpliwy I already did that but found same error

swaroopa94, 2018-03-13:
@Cierpliwy Yes I already did that but the error still exists

rishaChoubey, 2019-02-21:
I am facing the same issue when I'm running the code in updated version of Xcode 10.0 and the application is running in the lower version.