dotintent/react-native-ble-plx #233: npm install takes about 5 minutes to run

gre, 2018-03-08:
I have a RN project with bunch of native library but this library exclusively takes about ~3-5 minutes when running `npm install` the first time.
It looks like the native compilation is the reason of this (RxSwift complication, etc)
Is there a way libraries can be precompiled? like some sort of node-gyp prebuild equivalent? 🤔 

Thanks

Cierpliwy, 2018-03-12:
There is one thing which can be improved:

* By updating binary releases used by Carthage we end up only compiling BLE framework. All dependencies are cached.

Additionaly we can:
* Try to use static Swift framework to move compiling step to building process. That's how ObjC is working in most libraries.

* Wait for Swift ABI to stabilize and use prebuild frameworks. Unfortunately it will take a while.

Cierpliwy, 2018-03-27:
Another possibility is to reimplement API using just one Swift static library without any dependencies such as RxBluetoothKit.

pulkonet, 2018-06-27:
Use `yarn add`, it is faster than `npm install`.

gre, 2018-06-27:
@pulkonet i think yarn add is as slow as npm install regarding this problem. the same postinstall will run which is taking long time to compile. I actually was using yarn when i created this issue ;)

closed 2019-03-23 by stale[bot]