dotintent/react-native-ble-plx #234: What is the right way to reconnect to BLE when app is in background on IOS?

andycamp, 2018-03-08:
Our app requires us to maintain BLE connection regardless of whether app is in foreground or background. In IOS, when our app loses BLE connection while app is in background, it is not reconnecting until app is brought to the foreground again. We're trying to fix this, but are not having any luck.

We found some details on the what's going on on the IOS side: https://medium.com/arkulpa/ios-stay-connected-to-an-external-ble-device-as-much-as-possible-699d434846d2 and https://stackoverflow.com/questions/47409097/re-connect-ble-device-in-background-ios.  We're having real trouble translating this into working code using react-native-ble-plx. When we scan in the background, nothing happens, we're not even sure if it's scanning at all in the background.

Please let us know how to use this library to reconnect to a lost connection in the background on IOS.

pawelByszewski, 2018-03-09:
Could you provide more details about your implementation? and especially the information about which part of the reconnecting process does not start/work in the background. 
My first guess is that you do not use filtering by services during a scanning process (so the first parameter of `startDeviceScan` is empty or null), but it is just a guess.

Cierpliwy, 2018-03-13:
@andycamp Any news?

andycamp, 2018-03-13:
Thanks for asking! 

We used a packet sniffer to determine the service uuid wasn't broadcast in the advertised packets for the device we want to connect to. What we found is that we could not get startDeviceScan to work when passing the uuid for this device. 

As far as we found, this impacts IOS when attempting to reconnect to ble in the background. Android will allow us to do a new startDeviceScan in the background, find our device, discover services and characteristics, then continue working. IOS will not run a full scan again, and our device was not responding to the scan filtered by UUID. We have found that on IOS, we can simply maintain a reference to our device and just run device.connect directly without a new scan when we lose connection.

Is it a bluetooth requirement that service UUID's be advertised for the device? Or is that a limitation of this library?