dotintent/react-native-ble-plx #238: undefined is not an object (evaluating '_BleModule.BleModule.createClient')

ShunsakuVirus, 2018-03-16:
Hello!

I am trying to develop an application using BLE with react-native.

In developing it, since errors occurred many times, I consulted several sites, but none of them failed.

I tried referring to this site this time, but this time also failed.
[https://blog.expo.io/so-you-want-to-build-a-bluetooth-app-with-react-native-and-expo-6ea6a31a151d](url)

Please tell me how I can solve it.

My project was created with this command.
npx create-react-native-app myapp

next
cd myapp

exp login

npm install

exp detach

npm install - g react - native - cli

npm install react-native-ble-plx - save

react-native link react-native-ble-plx
In order, I updated my file like the above site.
Then,

exp start

Run and run on Expo on Android that is not USB connected to my computer, the following error occurs.
The content of my file is as follows.

myapp \ App.js:

`import React from 'react';
import { StyleSheet, Text, View } from 'react-native';
import { BleManager } from 'react-native-ble-plx';

export default class App extends React.Component {

  constructor() {
    super();
    this.manager = new BleManager();
  }

  render() {
    return (
      <View style={styles.container}>
        <Text>Open up App.js to start working on your app!</Text>
        <Text>Changes you make will automatically reload.</Text>
        <Text>Shake your phone to open the developer menu.</Text>
      </View>
    );
  }
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    backgroundColor: '#fff',
    alignItems: 'center',
    justifyContent: 'center',
  },
});
`

[... 493 lines cut ...]
  }

  // Needed for `react-native link`
  public List<ReactPackage> getPackages() {
    return Arrays.<ReactPackage>asList(
        // Add your own packages here!
        // TODO: add native modules!

        // Needed for `react-native link`
        // new MainReactPackage(),
            new BlePackage()
    );
  }

  @Override
  public String gcmSenderId() {
    return getString(R.string.gcm_defaultSenderId);
  }

  @Override
  public boolean shouldUseInternetKernel() {
    return BuildVariantConstants.USE_INTERNET_KERNEL;
  }
}
`

StephaneJuban, 2018-03-22:
What is the ouput of the command `react-native link react-native-ble-plx` ?
Because it looks like a lot like #49 or #93

Cierpliwy, 2018-04-08:
Closing as this is configuration error. Check above links for support.

closed 2018-04-08 by Cierpliwy