dotintent/react-native-ble-plx #240: Retrieving descriptors and properties from characteristics

patrick-samy, 2018-03-22:
Hello,

I'm trying to port a library for Android (provided by the manufacturer of my device) from Java to JS using react-native-ble-plx.

Specifically, I'm having trouble understanding how to enable notifications, retrieve descriptors and properties of a characteristic:
```
BluetoothGattService service = gatt.getService(AndroidBluetoothConnection.METER_UUID);
mBluetoothGattChar = service.getCharacteristic(AndroidBluetoothConnection.METER_UUID2);
mDescriptor = ((BluetoothGattDescriptor)mBluetoothGattChar.getDescriptors().get(0));
mCharaProp = mBluetoothGattChar.getProperties();
          
AndroidBluetoothConnection.this.setCharacteristicNotification(mBluetoothGattChar, true);
```

Interestingly, the characteristic discovered has "isNotifying" set to "false", even though the original code is enabling notification for the same characteristic. My code so far:
```
d.characteristicsForService('00001523-1212-efde-1523-785feabcd123').then((s) => {
    s.forEach((c) => {
        if (c.uuid === '00001524-1212-efde-1523-785feabcd123') {
            console.warn(c);
            const characteristic = d.writeCharacteristicWithResponseForService(
                c.serviceUUID,
                c.uuid,
                "AQ==" /* 0x01 in hex */
            ).then((characteristic) => {
                console.warn(characteristic);

                d.monitorCharacteristicForService(c.serviceUUID, c.uuid, (error, characteristic) => {
                /* Never reached */
                if (error) {
                    console.warn(error.message)
                    return;
                }
                console.warn('notified');
                console.warn(characteristic.uuid);
                console.warn(characteristic.value.toString());
            })
    })
    .catch((e) => {
        console.warn(e);
    })
});
```

Here is the library code for reference: https://gist.github.com/patrick-samy/4af06b06441e83fc9adb6b81db30ef3c

Cierpliwy, 2018-03-23:
> Specifically, I'm having trouble understanding how to enable notifications
https://polidea.github.io/react-native-ble-plx/#blemanagermonitorcharacteristicfordevice
Above function should enable notifications or indications.

> ,retrieve descriptors and properties of a characteristic:
You cannot directly get access to descriptors right know. However notifying and indicating should work.

> Interestingly, the characteristic discovered has "isNotifying" set to "false"
Characteristic object returned from API calls is always immutable. You need to get an instance
after setting notifications to be sure that notification was set or not. Currently there is no API to "refresh" characteristic. Alternatively you can set log level to verbose:
https://polidea.github.io/react-native-ble-plx/#blemanagersetloglevel
and check logcat to see what's going on.

Cierpliwy, 2018-04-08:
@patrick-samy Any news?

nixoschu, 2019-02-13:
Is there an Update on how to read / access the value of a Descriptor of a Characteristic ?

In my use case I have a ble sensor to read data (integer value) from. The Descriptor of the Characteristic is telling me how to scale the Integer (x1, x1/10, x1/100). 

I'm currently porting my app from ionic to react native and this seems to be the only good BLE Plugin on the board, so any help would be much appreciated.

Cierpliwy, 2019-04-05:
@nixoschu This is something, which is missing in the library and definitely should be implemented. I'll schedule work to fully support descriptors.

martin-gudmundson, 2019-06-06:
@Cierpliwy is there any timeframe planned for supporting descriptors?

Cierpliwy, 2019-06-07:
Implementation is in progress and some iOS code is present on `descriptors_support` branch. It definitely has the highest priority right now.

closed 2019-08-14 by stale[bot]

SuvabBaral, 2021-03-30:
Hi @Cierpliwy , the peripheral device I am using supports indicate, however, after the first command, my app(central device) is disabling the indication and all the subsequent commands are sent as notify. 
Is there a way I can continue sending commands as indicate rather than notify using the react-native-ble-plx?.

I have added the same question on stackoverflow: 
https://stackoverflow.com/questions/66868468/react-native-ble-plx-is-it-possible-for-configuring-indicate-as-a-standard-prot