dotintent/react-native-ble-plx #243: Upgrade swift version from 4.0.2 to 4.0.3 RxBluetoothKit and RxSwift

rugdealer, 2018-03-26:
Just attempted installation and saw this. No idea if this will have an effect later, just thought it should be noted.

> *** Skipped installing RxBluetoothKit.framework binary due to the error:
> 	"Incompatible Swift version - framework was built with 4.0.2 (swiftlang-900.0.69.2 clang-900.0.38) and the local version is 4.0.3 (swiftlang-900.0.74.1 clang-900.0.39.2)."
> *** Skipped installing RxSwift.framework binary due to the error:
> 	"Incompatible Swift version - framework was built with 4.0.2 (swiftlang-900.0.69.2 clang-900.0.38) and the local version is 4.0.3 (swiftlang-900.0.74.1 clang-900.0.39.2)."
>

Cierpliwy, 2018-03-27:
It's not an error. Binary frameworks provided by carthage were built  using Swift 4.0.2 and right now you have version 4.0.3. In that case, everything will be build from source and will take longer due to that. I'll try to update prebuilt binaries in 0.9.

wilr, 2018-04-16:
@Cierpliwy Any ETA on when the binaries will be updated?

radko93, 2018-04-25:
RxBluetoothKit seems to be updated for 9.3 now.

yogeshk19, 2018-04-30:
I get a similar error while running the npm install,

 `npm install --save react-native-ble-plx`

Here is the error I get.

> Skipped installing RxBluetoothKit.framework binary due to the error:
> 	"Incompatible Swift version - framework was built with 4.0.2 (swiftlang-900.0.69.2 clang-900.0.38) and the local version is 4.1 (swiftlang-902.0.48 clang-902.0.37.1)."
> *** Skipped installing RxSwift.framework binary due to the error:
> 	"Incompatible Swift version - framework was built with 4.0.2 (swiftlang-900.0.69.2 clang-900.0.38) and the local version is 4.1 (swiftlang-902.0.48 clang-902.0.37.1).

Thanks,
Yogesh

radko93, 2018-05-02:
@yogeshk19 This won't work until underlying library gets updated for new Swift.

Cierpliwy, 2018-05-25:
Latest version of RxSwift is still Spread.2 (https://github.com/ReactiveX/RxSwift/releases)

karimkhamwani, 2018-05-25:
@Cierpliwy when this issue get fixed thanks

ChrisEdson, 2018-06-09:
There’s a new RxSwift version released (‘Zurich’) which fixes this. Any chance of a new release now? We depend on this :)

Cierpliwy, 2018-06-12:
@ChrisEdson Please check version 0.9.1 if it's working for you.

efournival, 2018-06-19:
With the latest Xcode and `react-native-ble-plx` 0.9.1:
```
*** Checking out RxBluetoothKit at "4.0.3"
*** Checking out RxSwift at "4.2.0"
*** xcodebuild output can be found in /var/folders/hm/7dwsxrq964z38lgvsqzk6btc0000gn/T/carthage-xcodebuild.sHZbCr.log
*** Downloading RxSwift.framework binary at "Zürich"
*** Downloading RxBluetoothKit.framework binary at "4.0.3"
*** xcodebuild output can be found in /var/folders/hm/7dwsxrq964z38lgvsqzk6btc0000gn/T/carthage-xcodebuild.OhtlvM.log
*** Valid cache found for RxSwift, skipping build
*** Valid cache found for RxBluetoothKit, skipping build
*** Building scheme "BleClientManager" in BleClientManager.xcodeproj
Build Failed
	Task failed with exit code 6:
	/usr/bin/xcrun simctl list devices

This usually indicates that project itself failed to compile. Please check the xcodebuild log for more details: /var/folders/hm/7dwsxrq964z38lgvsqzk6btc0000gn/T/carthage-xcodebuild.OhtlvM.log
Error: "carthage build --no-skip-current --platform iOS --cache-builds"  command failed with status=1
```

Dina-Nashaat, 2019-01-15:
I think this problem still persists at my end, 
` Downloading RxBluetoothKit.framework binary at "4.0.3"
***  Skipped installing RxBluetoothKit.framework binary due to the error:
	"Incompatible Swift version - framework was built with 4.1.2 (swiftlang-902.0.54 clang-902.0.39.2) and the local version is 4.2.1 (swiftlang-1000.11.42 clang-1000.11.45.1)."

    Falling back to building from the source
***  Skipped installing RxSwift.framework binary due to the error:
	"Incompatible Swift version - framework was built with 4.1.2 (swiftlang-902.0.54 clang-902.0.39.2) and the local version is 4.2.1 (swiftlang-1000.11.42 clang-1000.11.45.1)."

    Falling back to building from the source`

which in turn gives me this when running the project:
`dyld: Library not loaded: @rpath/RxBluetoothKit.framework/RxBluetoothKit
  Referenced from: /Users/Dina/Library/Developer/Xcode/DerivedData/EpilarmApp-drlgussdbgntayfntobawjihvyrk/Build/Products/Debug-iphonesimulator/BleClientManager.framework/BleClientManager
  Reason: image not found`

Is anybody having the same problem, and if yes, how can I resolve it?

closed 2019-03-24 by stale[bot]