dotintent/react-native-ble-plx #244: serial communication phone to beacon

wowzaaa, 2018-03-28:
Hi

 I was testing out this lib in order to try to implement a serial communication between phones and a beacon
1) scanning connecting works
2) there are no services :(
3) these are no characteristics :(

i just need to send ABCDE ... the beacon will be programmed to do specific things based on the code it receives

so can i do this with this library ... i am kind of stuck right now :(

thanks

wowzaaa, 2018-03-29:
further details ... what i really need is bt transparent mode serial communication ... i cannot seem to find one library that does that :(

Cierpliwy, 2018-04-08:
What do you mean that there are no services and characteristics? Did you call `discoverServicesAndCharacteristics` function? What `nRF Connect` and similiar apps show when you connect to your device?

closed 2019-03-23 by stale[bot]

foundationx, 2020-11-13:
Did you figure this out?