dotintent/react-native-ble-plx #249: Can't find

AandersonL, 2018-04-08:
I can't find my device nearby my android device, i just find some devices thats dont have name, and almost everything is null. this library support find another android devices?

My code:

``` class Finder extends Component {
  constructor() {
    super()
    this.manager = new BleManager();
    this.manager.setLogLevel(LogLevel.Verbose)
    this.state = {status: "Scanner off",count:0, values: {}}
    this.devices = []
    this.startScan = this.startScan.bind(this);
    this.stopScan = this.stopScan.bind(this);
  }

  componentWillMount() {

  }

  startScan = () => {
    this.devices = [];
    this.setState({status: "Scanner on", count: 0});
    this.manager.startDeviceScan(null, null, (err, device) => {
      console.log("Scaning...");
      console.log(device);
      console.log(err);
      (device != null && device.name != null) ? devices.push(device.name) : "";
      if (err) console.log(err);
      this.setState({count:++this.state.count});

      device != null ? ToastAndroid.show(device.name, ToastAndroid.SHORT) : '';
    });
  }

  stopScan = () => {
    this.setState({status:"Scanner off"})
    this.manager.stopDeviceScan();
  }

Cierpliwy, 2018-04-09:
This library can find other Android only when it has an application which works in Bluetooth Low Energy Peripheral Mode. This library supports only Central Role.

AandersonL, 2018-04-15:
Just figure out i was looking for classics Bluetooths devices instede LE, my fault.

closed 2018-04-15 by AandersonL