dotintent/react-native-ble-plx #253: BleManager with Redux

karimraoudy, 2018-04-10:
Hello , i want to use ble manager in different component and the best way i think is with redux , but i 'dont know how to use it with BleManager() . thanks for your help

Cierpliwy, 2018-04-10:
You can create BleManager in redux-saga's generator which is then added to your redux as middleware. With redux-saga it is quite easy to handle asynchronous actions and emit new ones which will change your store and effectively your components.

karimraoudy, 2018-04-10:
thank you @Cierpliwy  for you're ansewr but can you give me some exemple to understand more

kevando, 2018-10-02:
@Cierpliwy  I am finally getting a grasp on redux-saga. I think it's going to help with alot of my issues. Can you provide a basic structure for how you would set up the ble manager with redux saga? Specifically to properly prepare for monitoring characteristics.

closed 2019-03-23 by stale[bot]

alejandropaciotti, 2019-04-21:
I don't know if you, @karimraoudy was able to get an example for understand more, but @Cierpliwy  did a perfect example.
https://github.com/Cierpliwy/SensorTag