dotintent/react-native-ble-plx #254: BLE standard services `assigned number`

jankoritak, 2018-04-10:
I have a BLE device, which runs bunch of standard BLE services, like for example a [blood pressure service](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.blood_pressure.xml). This service has a standard **assigned number** property, which it can be identified with.

For blood pressure service it is _0x1810_.

Does the package store this assigned number in corresponding service? Neither of services id or uuid properties seem to match it.

Cierpliwy, 2018-04-10:
`0x1810` is UUID written in short version. There are also 16bit UUID and full UUID types. Every UUID can be converted by filling remaining bits with specific pattern to bigger UUID. As we internally are forced to use strings to represent UUIDs it's hard to compare UUIDs logically. `00001810-0000-1000-8000-00805f9b34fb` is longer version of `1810` and can't be compared directly.

Therefore library outputs every UUID as full UUID with lowercase letters. If you want to convert to that format just `import { fullUUID } from 'react-native-ble-plx'` and use it to do conversion and then comparison:

`service.uuid == fullUUID("1810")`

jankoritak, 2018-04-10:
Makes sense, thanks!

closed 2018-04-10 by Cierpliwy