dotintent/react-native-ble-plx #255: android phone not found serviceUUIDs and Characteristics

engelsizmuzik, 2018-04-12:
not found android phone devices serviceUUIDs and Characteristics

```
list= () =>{
        console.log('list');
        this.manager.discoverAllServicesAndCharacteristicsForDevice(deviceid)
        .then((res)=>{
            console.log('ok');
            console.log(res);
        })
    }
```

> { isConnectable: null,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:   rssi: null,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:   serviceData: null,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:   overflowServiceUUIDs: null,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:   mtu: 23,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:   localName: null,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:   manufacturerData: null,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:   txPowerLevel: null,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:   id: 'FC:3D:93:30:8D:47',
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:   solicitedServiceUUIDs: null,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:   name: 'General',
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:   serviceUUIDs: null,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:   _manager:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:    { _eventEmitter:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:       { _subscriber:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:          { _subscriptionsForType:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:             { appStateDidChange:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                [ { subscriber: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    emitter:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                     { _subscriber: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                       isAvailable: true,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                       _eventHandlers:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                        { change:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                           { _mapData: [],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                             _objectIndex: {},
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                             _stringIndex: {},
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                             _otherIndex: {} },
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                          memoryWarning:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                           { _mapData: [],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                             _objectIndex: {},
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                             _stringIndex: {},
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                             _otherIndex: {} } },
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                       currentState: 'active' },
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    listener: [Function],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    context: undefined,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    eventType: 'appStateDidChange',
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    key: 0 } ],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:               websocketMessage: [ , , ,  ],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:               websocketOpen: [ , , ,  ],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:               websocketClosed: [ , , ,  ],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:               websocketFailed: [ , , ,  ],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:               didUpdateDimensions:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                [ { subscriber: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    emitter:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                     { _subscriber: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                       sharedSubscriber: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                       _currentSubscription: null },
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    listener: [Function],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    context: undefined,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    eventType: 'didUpdateDimensions',
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    key: 0 } ],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:               hardwareBackPress:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                [ { subscriber: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    emitter:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                     { _subscriber: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                       sharedSubscriber: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                       _currentSubscription: null },
[... 20 lines cut ...]
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    listener: [Function: collectExtraData],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    context: null,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    eventType: 'collectRedBoxExtraData',
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    key: 0 } ],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:               StateChangeEvent:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                [ { subscriber: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    emitter: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    listener: [Function],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    context: undefined,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    eventType: 'StateChangeEvent',
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    key: 0 } ],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:               toggleElementInspector:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                [ { subscriber: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    emitter:
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                     { _subscriber: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                       sharedSubscriber: [Circular],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                       _currentSubscription: null },
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    listener: [Function],
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    context: undefined,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    eventType: 'toggleElementInspector',
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:                    key: 0 } ] },
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:            _currentSubscription: null } },
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:      _uniqueId: 3,
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:      _activePromises: {},
> 04-12 11:40:17.909 13886 18343 I ReactNativeJS:      _activeSubscriptions: { '1': { remove: [Function: remove] } } } }

Cierpliwy, 2018-04-12:
`this.manager.discoverAllServicesAndCharacteristicsForDevice(deviceid)` returns device object. To get services use `services()` method:
https://polidea.github.io/react-native-ble-plx/#deviceservices

closed 2019-03-23 by stale[bot]

hidayatNgganteng, 2020-02-20:
i am change 

> device.discoverAllServicesAndCharacteristicsForDevice()

to 

> this.manager.discoverAllServicesAndCharacteristicsForDevice(peripheralId)

------------------------------------
and  change

> device.servicesForDevice()

to 

> this.manager.servicesForDevice(peripheralId)

------------------------------------

**and the problem is resolved**