dotintent/react-native-ble-plx #260: Help - How to return CBCentralManager to use in react-native-nordic-dfu

neomegas, 2018-04-17:
In [react-native-nordic-dfu](https://www.npmjs.com/package/react-native-nordic-dfu), there is a step to get CBCentalManager.   

>    Call [RNNordicDfu setCentralManagerGetter:<...>] with a block argument that returns 
    your CBCentralManager instance (see example project for how this may be done). 
    It is assumed that you have initiated this instance outside of this library, for flexibility.    

### the AppDelegate.m in the example (react-native-ble-manager is used)

     ...
    #import "RNNordicDfu.h"
    #import "BleManager.h"
    ...
    [RNNordicDfu setCentralManagerGetter:^() {
    return [BleManager getCentralManager];
    }];
    // Reset manager delegate since the Nordic DFU lib "steals" control over it
    [RNNordicDfu setOnDFUComplete:^() {
    NSLog(@"onDFUComplete");
    CBCentralManager * manager = [BleManager getCentralManager];
    manager.delegate = [BleManager getInstance];
    }];
    [RNNordicDfu setOnDFUError:^() {
    NSLog(@"onDFUError");
    CBCentralManager * manager = [BleManager getCentralManager];
    manager.delegate = [BleManager getInstance];
    }];   
    ...

### What I've tried (sadly, not work)

    ...
    #import "RNNordicDfu.h"
    @import BleClientManager;
    ...
    [RNNordicDfu setCentralManagerGetter:^() {
    return [BleClientManager init];
    }];
    ...

I need to make dfu work with ble-plx.
Does anybody know, how can I solve this?
My deps is
> "dependencies": {
    "immutable": "^3.8.2",
    "native-base": "^2.3.10",
    "react": "16.2.0",
    "react-native": "0.52.1",
    "react-native-ble-plx": "^0.9.0",
    "react-native-fs": "^2.9.12",
    "react-native-nordic-dfu": "^1.1.2",
    "react-native-slider": "^0.11.0",
    "react-native-splash-screen": "^3.0.6",
    "react-native-svg": "^6.1.3",
    "react-native-timer": "^1.3.1",
    "react-native-vector-icons": "^4.5.0",
    "react-navigation": "^1.4.0"
  },

Thank you for your help in advance.

neomegas, 2018-04-17:
I've succeed updating firmware on android using ble-plx with nordic-dfu. However, stucked in iOS

dariuszseweryn, 2018-05-09:
I was trying to use the `react-native-nordic-dfu` and had success just by using:

```
    [RNNordicDfu setCentralManagerGetter:^() {
        return [[CBCentralManager alloc] initWithDelegate:nil queue:dispatch_get_global_queue(DISPATCH_QUEUE_PRIORITY_BACKGROUND, 0)];
    }];
    
    // Reset manager delegate since the Nordic DFU lib "steals" control over it
    [RNNordicDfu setOnDFUComplete:^() {
        NSLog(@"onDFUComplete");
    }];
    [RNNordicDfu setOnDFUError:^() {
        NSLog(@"onDFUError");
    }];
```
And nothing more

closed 2019-03-23 by stale[bot]

RRMoelker, 2020-08-19:
> I was trying to use the `react-native-nordic-dfu` and had success just by using:
>
> ```
>     [RNNordicDfu setCentralManagerGetter:^() {
>         return [[CBCentralManager alloc] initWithDelegate:nil [...]

For reference, I've tried this. Even after connecting/scanning I still get that the the device id can not be found:

`[Error: Could not find device with deviceAddress]`

I've resorted to using the https://github.com/innoveit/react-native-ble-manager library which does work. But of course is less than ideal.

kinglaine, 2023-03-27:
> I've succeed updating firmware on android using ble-plx with nordic-dfu. However, stucked in iOS

How did you do it for android?

AshlandWest, 2023-05-02:
Has anyone had luck implementing this library on Android in React Native .70+ ?

niclaslindberg, 2023-08-31:
Yes, I'm also really interested if anyone got a working solution this on IOS?