dotintent/react-native-ble-plx #261: TypeError: undefined is not an object (evaluating '_BleModule.BleModule.createClient')

mshavandi, 2018-04-21:
I'm new to this library and whole react-native. I've done all steps written in this [site](https://blog.expo.io/so-you-want-to-build-a-bluetooth-app-with-react-native-and-expo-6ea6a31a151d)
url) successfully (It's the same as the steps in README.md file). To test this library, I used the code in the polidea site (this [Link](https://www.polidea.com/blog/ReactNative_and_Bluetooth_to_An_Other_level/)
but faced this error on my Android device (I'm working with a real device):

> TypeError: undefined is not an object (evaluating '_BleModule.BleModule.createClient')
> This error is located at:
> in SensorsComponent (at renderApplication.js:33)
> in RCTView (at View.js:60)
> in View (at AppContainer.js:102)
> in RCTView (at View.js:60)
> in View (at AppContainer.js:122)
> in AppContainer (at renderApplication.js:32)

This is my build.gradle file:

`apply plugin: "com.android.application"

import com.android.build.OutputFile

/**
 * The react.gradle file registers a task for each build variant (e.g. bundleDebugJsAndAssets
 * and bundleReleaseJsAndAssets).
 * These basically call `react-native bundle` with the correct arguments during the Android build
 * cycle. By default, bundleDebugJsAndAssets is skipped, as in debug/dev mode we prefer to load the
 * bundle directly from the development server. Below you can see all the possible configurations
 * and their defaults. If you decide to add a configuration block, make sure to add it before the
 * `apply from: "../../node_modules/react-native/react.gradle"` line.
 *
 * project.ext.react = [
 *   // the name of the generated asset file containing your JS bundle
 *   bundleAssetName: "index.android.bundle",
 *
 *   // the entry file for bundle generation
 *   entryFile: "index.android.js",
 *
 *   // whether to bundle JS and assets in debug mode
 *   bundleInDebug: false,
 *
 *   // whether to bundle JS and assets in release mode
 *   bundleInRelease: true,
 *
 *   // whether to bundle JS and assets in another build variant (if configured).
 *   // See http://tools.android.com/tech-docs/new-build-system/user-guide#TOC-Build-Variants
 *   // The configuration property can be in the following formats
 *   //         'bundleIn${productFlavor}${buildType}'
 *   //         'bundleIn${buildType}'
 *   // bundleInFreeDebug: true,
 *   // bundleInPaidRelease: true,
 *   // bundleInBeta: true,
 *
 *   // whether to disable dev mode in custom build variants (by default only disabled in release)
 *   // for example: to disable dev mode in the staging build type (if configured)
 *   devDisabledInStaging: true,
 *   // The configuration property can be in the following formats
 *   //         'devDisabledIn${productFlavor}${buildType}'
 *   //         'devDisabledIn${buildType}'
 *
 *   // the root of your project, i.e. where "package.json" lives
 *   root: "../../",
 *
 *   // where to put the JS bundle asset in debug mode
 *   jsBundleDirDebug: "$buildDir/intermediates/assets/debug",
 *
 *   // where to put the JS bundle asset in release mode
 *   jsBundleDirRelease: "$buildDir/intermediates/assets/release",
 *
 *   // where to put drawable resources / React Native assets, e.g. the ones you use via
 *   // require('./image.png')), in debug mode
 *   resourcesDirDebug: "$buildDir/intermediates/res/merged/debug",
 *
[... 158 lines cut ...]
    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
          new MainReactPackage()
      );
    }

    @Override
    protected String getJSMainModuleName() {
      return "index";
    }
  };

  @Override
  public ReactNativeHost getReactNativeHost() {
    return mReactNativeHost;
  }

  @Override
  public void onCreate() {
    super.onCreate();
    SoLoader.init(this, /* native exopackage */ false);
  }
}`

How to solve this problem? None of the solutions in the google search was my case.

mshavandi, 2018-04-22:
Just hadn't done steps 4 to 6 of the manual setup of react-native-ble-plx. refer to this [link](https://github.com/Polidea/react-native-ble-plx#android).

closed 2018-04-22 by mshavandi

nielsswinkels, 2018-06-20:
Same thing also for android projects. Follow the steps https://github.com/Polidea/react-native-ble-plx#android

nicoache1, 2018-09-25:
You need to add this lines in your main application

import com.polidea.reactnativeble.BlePackage;

and this

    @Override
    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
          new MainReactPackage(),
            new BlePackage()
      );
    }

thefakhreddin, 2018-10-17:
try react-native link `react-native-ble-plx` not  just `react-native link`

thefakhreddin, 2018-10-17:
solve this...............
try `react-native link react-native-ble-plx` not just `react-native link`