dotintent/react-native-ble-plx #262: device.name always null when scanning

Pa-well, 2018-04-24:
Hi!
For several days I try to connect to any BT device. First, I was building my application using https://www.polidea.com/blog/ReactNative_and_Bluetooth_to_An_Other_level/, but I could not connect to my device. I've found out that program stops in scanAndConnect() function, because no device name fits to the one that I am looking for. I've used function like this to find that devices can be detected:

scanAndConnect() {
      let device;
        let x = 0;
      this.manager.startDeviceScan(null, null, (error, device) => {
        console.log(x);
        console.log(error);
        console.log(device.name);
        x = x+1;
      });
    }

but the result is always like this:
I/ReactNativeJS(16857): 5882
I/ReactNativeJS(16857): null
I/ReactNativeJS(16857): null
I/ReactNativeJS(16857): 5883
I/ReactNativeJS(16857): null
I/ReactNativeJS(16857): null
etc.

my code is the one from issue #83 proposed by @Cierpliwy, I have put everything in my github repo: https://github.com/Pa-well/BtTest

My device is Arduino HC-05 BT modul, and I can connect to it from other applications, so it works. Devices are paired. Am I missing something obvious?

closed 2018-04-24 by Pa-well

Pa-well, 2018-04-24:
Ok, I have found that this lib supports only BLE devices,
That was really obvious :)

RameshAmzur, 2019-05-28:
Hi @pawel125 
I have tested whether   my device supports BLE or not.It supports BLE but its name is not showing.

Thank you.