dotintent/react-native-ble-plx #264: Error at monitorCharacteristicForDevice

bvic23, 2018-04-25:
Hello There,

I love your project! I have a small issue, cannot set monitoring on: whenever I call `monitorCharacteristicForDevice` I get the following error:

` Error: Characteristic 92eeb32f-470c-0074-fd63-39171798c304 notify change failed for device ? and service 83443868-d0cc-d599-1d9d-e23a1911a98b`

[image]

Any tips & tricks are welcome!

Thanks!

JanErikFoss, 2018-07-18:
Me too, what's the status on this?

kevando, 2018-10-02:
@bvic23 try reading the characteristic before you monitor. This allows me to get data in the monitor function, but its still wacky.

bvic23, 2018-10-02:
👍

vaukalak, 2018-10-25:
@kevando you mean running 
```
await characteristic.read();
```
before
```
characteristic.monitor(() => {});
```
?
That actually didn't help, still getting same error.

ThomasStubbe, 2018-10-31:
I'm having the same issue. @kevando Do you mean `discoverAllServicesAndCharacteristicsForDevice` ? I do this before I start monitoring, but the error still occurs

dikamilio, 2019-01-02:
I have the same problem even with reading characteristic before monitoring.
Digging deeper in error I fount following cause:
`Cannot find client characteristic config descriptor (code 2) with characteristic UUID ... `

bhaveshj-cuelogic, 2019-02-28:
I am having the same issue in iOS plz suggest some solution

Cierpliwy, 2019-04-05:
@dariuszseweryn Is it something, which should work now? I think we modified source code related to notification registration on Android.

dariuszseweryn, 2019-04-05:
Yes, it should be fixed since [`1.0.2`](https://github.com/Polidea/react-native-ble-plx/releases/tag/1.0.2)

closed 2019-04-05 by dariuszseweryn

packetstracer, 2019-05-21:
Happening to me on iOS with library version 1.0.3

This the stack trace that the error returns 

`"BleError: Characteristic d8b2c95b-b317-47ca-ac02-ccfd3d85a666 notify change failed for device 80293EA3-F380-A86F-B84F-DFC1A5B23502 and service ef66ccad-f4a8-4c51-b025-bc2a2c9fdaa6
    at parseBleError (blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:112674:18)
    at monitorListener (blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:113739:50)
    at RCTDeviceEventEmitter.emit (blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:6131:37)
    at MessageQueue.__callFunction (blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:2587:44)
    at blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:2344:17
    at MessageQueue.__guard (blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:2541:13)
    at MessageQueue.callFunctionReturnFlushedQueue (blob:http://10.80.148.183:8081/e724eaee-ffd5-4243-9d1e-3c4c983e62ea:2343:14)
    at http://10.80.148.183:8081/debugger-ui/debuggerWorker.js:80:58"`

The characteristic works fine, I've tested It with BLE Light app.

packetstracer, 2019-05-22:
happening to me on iOS, versions: 1.0.2 and 1.0.3

opened an issue

[https://github.com/Polidea/react-native-ble-plx/issues/468](issue)

marquezbwwt, 2021-02-24:
Is anyone else still getting this error and did you find a solution?  I'm seeing the same thing and am using v2.0.2.