dotintent/react-native-ble-plx #265: [iOS] Device.onDisconnected() emits nulls

dariuszseweryn, 2018-04-26:
When running the below code snippet the error is a `null` instead of `BleError`. It is observed on `iOS` only.
```
  device.onDisconnected((error) => {
    console.log(typeof error); // 'object'
    console.log(error === null); // 'true'
  });
```
On Android the library seems to work just fine.

dominikeinkemmer, 2018-06-08:
The checks you log are the default javascript behavior it seems (screenshot from chrome console).
[image]

I assume that the library emits null for the error variable if there is no error during disconnecting. So if you want to check if there was an error and do something with it, you can just check for `if(error)` .

dariuszseweryn, 2018-06-08:
I have induced disconnection from the outside—I would expect the library to consistently report this situation as an exception (on Android it behaves this way)

suark, 2018-10-10:
I am also seeing this behavior. 

1. Connect to a BLE device. 
2. Connect to that BLE in our app.
3. Unplug the BLE device.
4. a. On android it will trigger device.onDisconnected with error defined as a BleError.
4. b. On iOS it will trigger device.onDisconnected with the error === null

suark, 2018-10-11:
@Cierpliwy I feel confused because those docs say "If an error is null, that means the connection was terminated by bleManager.cancelDeviceConnection() call.". 

To me, this implies that a "null" error means an intentional disconnection. If my ble server suddenly loses power, it was not an intentional disconnection. Yet it seems impossible to determine the difference between this and when we purposefully call the function(cancelDeviceConnection()) to manually disconnect from a device. 

Also you mention "On iOS, when connection was terminated by an application error might be set to null." but the docs don't mention any special case for iOS in your link. 
[image]

To me, the problem is that the behavior is inconsistent between android and iOS. On android I seem to be able to reliably count on have a defined error when the disconnection is caused due to an error, such as suddenly losing connection to a ble server device. But on iOS it seems that it's impossible to distinguish this difference so we are unable to know if our user clicked "disconnect" or the device just disappeared off the face of the earth. 

How are we supposed to detect that we just lost connection to the BLE server device if the device.onDisconnected function is being passed a null error for iOS, but a defined error for android?

suark, 2018-10-17:
@Cierpliwy Is there any way we could detect this and throw an error so that the behavior is the same for android and ios for react-native users?

closed 2019-03-23 by stale[bot]

dennis-b, 2019-06-04:
i also seeing same behavior on ios, is there any update on this one?

cernadasjuan, 2019-08-29:
@suark Thanks, I implemented what do you say and it's working fine

suark, 2019-08-29:
@cernadasjuan that's awesome. Cheers!