dotintent/react-native-ble-plx #271: [Android] Error: Device ? is already connected

jankoritak, 2018-05-11:
I'm using following, very simple, piece of boilerplate code.

```

...

manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        console.warn(error)
        return
      }

      if (device.name === this.DEVICE_NAME) {
         this.connectTo(device, false)
      }
    })

...

const connectTo = (device, isLeft) => {
   device.connect()
      .then((device) => {
         console.log("[Info][connectTo][success]", device)
      })
      .catch((error) => {
         console.log('[Error][connectTo][error]: ', error)
      })
}

... 
```

The code runs as expected on iOS devices, but fails on Android devices.

Every-time I try to connect a device, I'm getting `BleErrorCode.DeviceAlreadyConnected` error, viz.

```
[Error][connectTo]:  Error: Device ? is already connected
    at new BleError (blob:http://192.168.86.39:19001/fed3bf61-e367-4015-871f-d29b37905b0e:114897:110)
    at parseBleError (blob:http://192.168.86.39:19001/fed3bf61-e367-4015-871f-d29b37905b0e:114917:18)
    at BleManager._callPromise$ (blob:http://192.168.86.39:19001/fed3bf61-e367-4015-871f-d29b37905b0e:115168:51)
    at tryCatch (blob:http://192.168.86.39:19001/fed3bf61-e367-4015-871f-d29b37905b0e:17030:19)
    at Generator.invoke [as _invoke] (blob:http://192.168.86.39:19001/fed3bf61-e367-4015-871f-d29b37905b0e:17203:24)
    at Generator.prototype.(anonymous function) [as throw] (blob:http://192.168.86.39:19001/fed3bf61-e367-4015-871f-d29b37905b0e:17073:23)
    at tryCatch (blob:http://192.168.86.39:19001/fed3bf61-e367-4015-871f-d29b37905b0e:17030:19)
    at invoke (blob:http://192.168.86.39:19001/fed3bf61-e367-4015-871f-d29b37905b0e:17106:22)
    at blob:http://192.168.86.39:19001/fed3bf61-e367-4015-871f-d29b37905b0e:17118:15
    at tryCallOne (blob:http://192.168.86.39:19001/fed3bf61-e367-4015-871f-d29b37905b0e:16359:14)

[Error][connectTo]:  Error: Operation was cancelled
    at new BleError (blob:http://192.168.86.39:19001/d1e699a5-985a-40dc-a7a0-7b2a4f49e47a:114897:110)
    at parseBleError (blob:http://192.168.86.39:19001/d1e699a5-985a-40dc-a7a0-7b2a4f49e47a:114917:18)
    at BleManager._callPromise$ (blob:http://192.168.86.39:19001/d1e699a5-985a-40dc-a7a0-7b2a4f49e47a:115168:51)
    at tryCatch (blob:http://192.168.86.39:19001/d1e699a5-985a-40dc-a7a0-7b2a4f49e47a:17030:19)
    at Generator.invoke [as _invoke] (blob:http://192.168.86.39:19001/d1e699a5-985a-40dc-a7a0-7b2a4f49e47a:17203:24)
    at Generator.prototype.(anonymous function) [as throw] (blob:http://192.168.86.39:19001/d1e699a5-985a-40dc-a7a0-7b2a4f49e47a:17073:23)
    at tryCatch (blob:http://192.168.86.39:19001/d1e699a5-985a-40dc-a7a0-7b2a4f49e47a:17030:19)
    at invoke (blob:http://192.168.86.39:19001/d1e699a5-985a-40dc-a7a0-7b2a4f49e47a:17106:22)
    at blob:http://192.168.86.39:19001/d1e699a5-985a-40dc-a7a0-7b2a4f49e47a:17118:15
    at tryCallOne (blob:http://192.168.86.39:19001/d1e699a5-985a-40dc-a7a0-7b2a4f49e47a:16359:14)
```

This sequence basically repeats until somehow devices reconnects on it's own. I don't handle reconnection from my side.

On the other hand, when I try to call

```
device.isConncted()
```

I'm getting `false`.

As well as when I try to force device disconnect before I try connecting it back again via

```
device.cancelConnection()
```

I'm getting `BleErrorCode.DeviceNotConnected`

Is this a bug? Any suggestions how to bypass this for now?

burkaslarry, 2018-05-19:
no official  developer answers this , it seems.

dariuszseweryn, 2018-05-19:
@jankoritak I do not see that you stop scanning at the time of the first scan of your device. So you esentially try to connect each time the device is scanned. And connection takes time. If the device is not yet connected it is advertising. Therefore if you try to establish more than one connection you get the `BleErrorCode.DeviceAlreadyConnected`.

Best Regards

closed 2018-05-19 by dariuszseweryn

lastmjs, 2018-05-25:
It seems there is no easy way to tell if the device is connected though. It would be ideal to write code like the following:

```javascript
manager.startDeviceScan(null, null, async (error, device) => {
  if (device.name === 'THE_DEVICE') {
    const deviceConnected = await device.isConnected();

    if (deviceConnected) {
      await device.cancelConnection();
    }

    device.connect();

     // do the stuff that you want to do
  }
});
```

I don't understand why the above code will not work, it doesn't on my Android device. I'm guarding against multiple connections, but some timing or something with the asynchronous nature of these function calls seems off. Any help would be greatly appreciated, amazing library.

kazuooooo, 2018-06-13:
/**
 * Sample React Native App
 * https://github.com/facebook/react-native
 * @flow
 */

import React, { Component } from 'react';
import {
  Alert,
  Platform,
  StyleSheet,
  Text,
  View,
  Button
} from 'react-native';
import { BleManager } from 'react-native-ble-plx';
import { Table, Row, Rows } from 'react-native-table-component';

const instructions = Platform.select({
  ios: 'Press Cmd+R to reload,\n' +
    'Cmd+D or shake for dev menu',
  android: 'Double tap R on your keyboard to reload,\n' +
    'Shake or press menu button for dev menu',
});

type Props = {};

const targetDeviceId = '24:0A:C4:0F:B5:42'
export default class App extends Component<Props> {

  constructor(props) {
    super(props);
    this.state = {
      scanning: false,
      findBeacons: [],
      tableHead: ['ID', 'NAME', 'RSSI', 'SERVICE UUID'],
      tableData: []
    };
    this.manager = new BleManager();
    this.device = null
    this.ring             = this.ring.bind(this)
    this.cancelConnection = this.cancelConnection.bind(this)
    this.checkConnection  = this.checkConnection.bind(this)
    this.connect          = this.connect.bind(this)
  }

  componentWillMount() {
    const subscription = this.manager.onStateChange((state) => {
      if (state === 'PoweredOn') {
        console.log("startScan")
        this.scanAndConnect();
        subscription.remove();
      }
    }, true);
  }

  scanAndConnect() {
    this.manager.startDeviceScan(null, null, (error, device) => {
      if (error) {
        console.log(`SCAN_ERROR: ${error}`)
        // Handle error (scanning will be stopped automatically)
        return
      }
      // Check if it is a device you are looking for based on advertisement data
      // or other criteria.
      // console.log(JSON.stringify(device))
      const row =  [device.id, device.name, device.rssi, device.serviceUUIDs]

      if(device.id === targetDeviceId && this.device === null) {
        this.device = device

        // this.connectTo(device)
        this.setState(prevState => (
          {
            tableData: [row, ...prevState.tableData,]
          }
        ))
      }

      if (device.name === 'TI BLE Sensor Tag' ||
        device.name === 'SensorTag') {

        // Stop scanning as it's not necessary if you are scanning for one device.
        this.manager.stopDeviceScan();

        // Proceed with connection.
      }
    });
  }

  ring(){
    this.manager.writeCharacteristicWithResponseForDevice(
      this.device.id,
      '8A17CFBC-83BB-4288-A850-51FCE4842CA8',
      'BEB5483E-36E1-4688-B7F5-EA07361B26A8',
      'YWJjZGVmZw=='
    ).then((res) => {
      console.log(`WRITE RES ${res}`)
    }).catch((error) => {
      console.log(`WRITE ERROR ${error}`)
    })
  }

  connect(){
    this.device.connect()
      .then((dev) => {
        console.log("try to connect")
        return dev.discoverAllServicesAndCharacteristics()
      })
      .then((dev) => {
        console.log("success to connect")
      })
      .catch((error) => {
        console.log("connect error:" + error)
      });
  }

  cancelConnection(){
    this.device.cancelConnection()
      .then((res) => {
        console.log("connection cancel")
        console.log(res)
      })
      .catch((err) => {
        console.log("connection cancel error:" + err)
      })
  }

  checkConnection(){
    this.device.isConnected()
      .then((res) => {
        console.log("connection check:" + res)
      })
      .catch((err) => {
        console.log("connection check:" + err)
      })
  }

  render() {
    return (
      <View style={styles.container}>
        <Table borderStyle={{borderWidth: 2, borderColor: '#c8e1ff'}}>
          <Row data={this.state.tableHead} style={styles.head} textStyle={styles.text}/>
          <Rows data={this.state.tableData} textStyle={styles.text}/>
        </Table>
        <Button title="Ring" onPress={this.ring}></Button>
        <Button title="Connect" onPress={this.connect}></Button>
        <Button title="Cancel Connection" onPress={this.cancelConnection}></Button>
        <Button title="Check Connection" onPress={this.checkConnection}></Button>
      </View>
    );
  }
}

const styles = StyleSheet.create({
  container: { flex: 1, padding: 16, paddingTop: 30, backgroundColor: '#fff' },
  head: { height: 40, backgroundColor: '#f1f8ff' },
  text: { margin: 6 }
});

NdozHH, 2018-08-20:
@kazuooooo Do you know how to trigger the notification sound of the BLE device?

asurare, 2019-07-26:
I don't really understand why this this issue is closed... Any proper way to fix that error ?

Thanhcao86, 2019-11-12:
id of Device not match.

tomaks-moon, 2020-07-24:
> id of Device not match.

What do you mean?

HualiangLI, 2020-12-15:
You may check the connection before connect the device.

```js
    device
      .isConnected() // check whether it's already conncted
      .then((isConnected) => {
        console.log('>>>', isConnected);
        return isConnected ? device : device.connect();
      })
      .then((d) => {
        console.log('connect::', d, d.serviceData, d.manufacturerData);
        return d.discoverAllServicesAndCharacteristics();
      })
      .then((d) => {
        console.log('discoverAllServicesAndCharacteristics::', d);
        return d.services();
      })
```

shreyakapoor072, 2022-10-28:
@HualiangLI 
I tried your solution and also ones mentioned above. But the problem is promise is not resolving successfully. Basically not moving to discoverAllServicesAndCharacteristics step, hence connection is not being established. Also isConnected is also false only most of the times. After some time if your device ( weighing scale in my case) gets powered off then it goes in catch block and gives disconnected error. The problem is with success resolution. Can anyone plz help in this ?

Simeonhm, 2024-05-04:
Anyone have a proper solution?

I got this:

 ERROR  [BleError: Device ? is already connected]
 ERROR  [BleError: Operation was cancelled]

MaxTheTurtle0, 2025-03-09:
I sometimes run into the same issue.

olalonde, 2025-04-07:
Also running into this issue. Fairly sure it's either a bug with Android or this library (probably Android).