dotintent/react-native-ble-plx #273: Cannot find bluetooth enabled Raspberry Pi

geojins, 2018-05-13:
Hi,

Apologies if this is too dump question, I am a beginner. 

I am trying to connect to a Raspberry Pi 3 (  Bluetooth 4.2, BLE) from an android app using 0.9.0 of  react-native-ble-plx from Moto 3G.  Issue I am facing is that the startDeviceScan does not return any valid devices.  After waiting for some time, a device is detected, but seems be some junk, see below the device printed. It is definitely not my raspberry pi. Can you shed some light?. My code snippet is given below
( If I use the android bluetooth search from phone settings, I can see the raspberry pi and connect to it.)

```
{ isConnectable: null,
   rssi: -99,
   serviceData: null,
   overflowServiceUUIDs: null,
   mtu: 23,
   localName: null,
   manufacturerData: 'TAAMDgDfuhsMDY5OnfIxmU6xEAULECrULw==',
   txPowerLevel: null,
   id: '56:B1:6D:8C:60:C8',
   solicitedServiceUUIDs: null,
   name: null,
   serviceUUIDs: null,
   _manager: 
    { _eventEmitter: 
       { _subscriber: 
          { _subscriptionsForType: 
             { appStateDidChange: 
                [ { subscriber: [Circular],
                    emitter: 
                     { _subscriber: [Circular],
                       isAvailable: true,
                       _eventHandlers: 
                        { change: 
                           { _mapData: [],
                             _objectIndex: {},
                             _stringIndex: {},
                             _otherIndex: {} },
                          memoryWarning: 
                           { _mapData: [],
                             _objectIndex: {},
                             _stringIndex: {},
                             _otherIndex: {} } },
                       currentState: 'active' },
                    listener: [Function],
                    context: undefined,
                    eventType: 'appStateDidChange',
                    key: 0 } ],
               websocketMessage: [ , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,  ],
               websocketOpen: [ , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,  ],
               websocketClosed: [ , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,  ],
               websocketFailed: [ , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , , ,  ],
               didUpdateDimensions: 
                [ { subscriber: [Circular],
                    emitter: 
                     { _subscriber: [Circular],
                       sharedSubscriber: [Circular],
                       _currentSubscription: 
                        { subscriber: [Circular],
                          emitter: [Circular],
                          listener: [Function: scanListener],
                          context: undefined,
                          eventType: 'ScanEvent',
                          key: 0 } },
                    listener: [Function],
                    context: undefined,
                    eventType: 'didUpdateDimensions',
                    key: 0 } ],
               hardwareBackPress: 
                [ { subscriber: [Circular],
                    emitter: 
                     { _subscriber: [Circular],
                       sharedSubscriber: [Circular],
                       _currentSubscription: 
                        { subscriber: [Circular],
                          emitter: [Circular],
                          listener: [Function: scanListener],
                          context: undefined,
                          eventType: 'ScanEvent',
                          key: 0 } },
                    listener: [Function],
                    context: undefined,
                    eventType: 'hardwareBackPress',
                    key: 0 } ],
               collectBugExtraData: 
                [ { subscriber: [Circular],
                    emitter: 
                     { _subscriber: [Circular],
                       sharedSubscriber: [Circular],
                       _currentSubscription: 
                        { subscriber: [Circular],
                          emitter: [Circular],
                          listener: [Function: scanListener],
                          context: undefined,
                          eventType: 'ScanEvent',
                          key: 0 } },
                    listener: [Function: collectExtraData],
                    context: null,
                    eventType: 'collectBugExtraData',
                    key: 0 } ],
               collectRedBoxExtraData: 
                [ { subscriber: [Circular],
                    emitter: 
                     { _subscriber: [Circular],
                       sharedSubscriber: [Circular],
                       _currentSubscr
```

Here is my code:
```
export default class App extends Component<Props> {
  constructor() {
    super();
    this.manager = new BleManager();
}

componentWillMount() {
    console.log("in componentWillMount")
    const subscription = this.manager.onStateChange((state) => {
        if (state === 'PoweredOn') {
            this.scanAndConnect();
            subscription.remove();
        }
    }, true);
}

scanAndConnect() {
  console.log("in scanAndConnect")
    this.manager.startDeviceScan(null, null, (error, device) => {
        if (error) {
            // Handle error (scanning will be stopped automatically)
            console.log("ERROR : ", error);
            return
        }

        console.log("Device Detected", device);
        // Check if it is a device you are looking for based on advertisement data
        // or other criteria.
        if (device.name === 'TI BLE Sensor Tag' ||
            device.name === 'SensorTag') {

            // Stop scanning as it's not necessary if you are scanning for one device.
            this.manager.stopDeviceScan();

            // Proceed with connection.
        }
    });
}

render() {
    return (
      <View style={styles.container}>
        <Text style={styles.welcome}>
          Welcome to React Native!
        </Text>
        <Text style={styles.instructions}>
          To get started, edit App.js
        </Text>
        <Text style={styles.instructions}>
          {instructions}
        </Text>
      </View>
    );
  }
}
```

geojins, 2018-05-15:
Found an answer to my own question. The issue was raspberry pi  not being setup as a peripheral by default. 
If anyone runs into this issue, refer [this](https://tobiastrumm.de/2016/10/04/turning-a-raspberry-pi-3-into-a-bluetooth-low-energy-peripheral/) for making  a raspberry pi as peripheral and advertise it

closed 2018-05-15 by geojins

andiskim, 2019-11-25:
@geojins I am having the same issue but the link you provided does not work. How can I resolve this issue?